microsoft/AirSim #260: Understanding Telemetry communication with RC and PixHawk-AirSim communication

nickylimjj, 2017-06-06:
Hi, how does the RC-data get relayed from the pixhawk to the MavLinkDroneController object in Unreal? Also, if I am not wrong, MavLinkDroneController is the virtualization of the flight controller? How does it then work in tandem with the physical hardware (ie, PixHawk)? From my understanding, the MavLinkDroneController processes MavLink Messages in MavlinkDroneControllerImpl.cpp` which is generated data from the AirSim sensors library and physics engine.

Thanks for any helping!

nickylimjj, 2017-06-09:
In addition, is anyone familiar with the difference between the `MavlinkDroneController` and `mavlinkVehicle` objects?

lovettchris, 2017-06-09:
MavLinkCom library, which includes MavlinkVehicle provides a lower level communications channel for talking to mavlink based drones, including the PX4 based Pixhawk.  For information on this library see the [readme](https://github.com/Microsoft/AirSim/blob/master/MavLinkCom/README.md).

AirSim uses this library, and you correctly identified the class that bridges the AirSim DroneControlBase API to MavLink which is MavLinkDroneController.  AirSim then simulated the physics and the drone sensors (IMU, GPS, Baro, etc) and sends that data over mavlink to the PX4 drone.  The drone responds to that sensor data by sending motor control messages, which are then intercepted by AirSim and sent to the Physics engine which turns that into thrust on the physics model of the drone, which is then visualized using the Unreal game engine.  Therein lies the entire closed loop which results in a complete software in the loop (SITL) or hardware in the loop (HITL) drone simulation.

nickylimjj, 2017-06-12:
Thank you chris! I have been experimenting with your programs -- namely mavlinktest and logviewer -- to understand it. Those are great and well written programs!

I am trying to craft my own mavlink packets to learn the protocol better, but am unsure how to get it transmitted. I tried modifying code in `MavLinkConnectionImpl` `sendMessage` function and `MavLinkDroneController` `processMavMessage` function but remain stuck. Are you able to provide any tips?

Thank you!

lovettchris, 2017-06-12:
MavLinkTelemetry is an example of a custom mavlink message, you can see the code in there, but the more general solution is to add to mavlink [common.xml](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml) and run the [MavLinkComGenerator](https://github.com/Microsoft/AirSim/tree/master/MavLinkCom/MavLinkComGenerator) to regenerate these files: [MavLinkMessages.hpp](https://github.com/Microsoft/AirSim/blob/master/MavLinkCom/include/MavLinkMessages.hpp) and [MavLinkMessages.cpp](https://github.com/Microsoft/AirSim/blob/master/MavLinkCom/src/MavLinkMessages.cpp).

closed 2017-06-13 by nickylimjj

Mahdi-s, 2020-10-05:
I just wanted to say thank you, Nicky and Chris, this was a great thread!