microsoft/AirSim #262: Flying Drone In A Circle With the Camera Pointed at the Center

exploke, 2017-06-06:
I am attempting to take pictures of an object from 360 degrees. Doing this with the RC is very difficult to do, but I found that there is code to do this in the AirSim/MavLinkCom/README.md under Examples. Does somebody know how I can execute this code?

clovett, 2017-06-06:
You don't need MavLinkTest, that is just a unit test app.  But instead you can do this using DroneShell to create an orbit that keeps the drone pointed at the center of the circle:

==||=> circlebypath -radius 5 -drivetrain 1 -yaw 90 -velocity 1 -z -5

This is perfect for creating 3D scans.

vatsagandhi, 2017-06-06:
==||=> circlebypath -radius 5 -drivetrain 1 -yaw 90 -velocity 1 -z -5
==||=> Async RPC Error: rpclib: client error C0002: Function 'moveOnPath' was called with an invalid number of arguments. Expected: 7, got: 6

The drone shell is saying I do not have enough arguments. I even tried putting the duration parameter in, but I still go the same error from drone shell.

clovett, 2017-06-06:
this means your unreal environment is using incompatible AirSim bits.  See the steps listed in [unreal_blocks.md](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md).

vatsagandhi, 2017-06-07:
It works thank you. What are the units for when we specify radius and velocity? Is it m and m/s respectively?

lovettchris, 2017-06-07:
yes, all distances are in meters, and all velocities are in meters per second.

vatsagandhi, 2017-06-07:
Thank you for your help. I am closing this thread.

closed 2017-06-08 by exploke

sytelus, 2018-04-12:
We had a bug that caused yawing unnaturally for the drone. It's fixed now. There is orbit.py in PythonClient by @lovettchris which does circle while pointing at origin.