microsoft/AirSim #265: UE4 crashes when I click play for Hardware in Loop

smishraasu, 2017-06-08:
I have a Pixhawk connected and when I click the play button in my UE4 environment I get the following error : 

LoginId:58ecec83694caf436454ebad3cf2c9a4
EpicAccountId:563b9e126ed8417884c542d2bc820d15

 #0 at 0x7fffb7b7bd42

__pthread_kill Address = 0x7fffb7b7bd42 (filename not found) [in libsystem_kernel.dylib]
abort Address = 0x7fffb7ae1420 (filename not found) [in libsystem_c.dylib]
__cxa_bad_cast Address = 0x7fffb663784a (filename not found) [in libc++abi.dylib]
default_terminate_handler() Address = 0x7fffb665cc37 (filename not found) [in libc++abi.dylib]
Unknown() Address = 0x7fffb716a713 (filename not found) [in libobjc.A.dylib]
std::__terminate(void (*)()) Address = 0x7fffb6659d69 (filename not found) [in libc++abi.dylib]
__cxxabiv1::exception_cleanup_func(_Unwind_Reason_Code, _Unwind_Exception*) Address = 0x7fffb66597de (filename not found) [in libc++abi.dylib]
msr::airlib::MavLinkDroneController::impl::createMavSerialConnection(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, int) Address = 0x17e9d5c71 (filename not found) [in UE4Editor-AirSim.dylib]
msr::airlib::MavLinkDroneController::impl::createMavConnection(msr::airlib::MavLinkDroneController::ConnectionInfo const&) Address = 0x17e9d5410 (filename not found) [in UE4Editor-AirSim.dylib]
msr::airlib::MavLinkDroneController::impl::connect() Address = 0x17e9d3dc5 (filename not found) [in UE4Editor-AirSim.dylib]
msr::airlib::MavLinkDroneController::impl::start() Address = 0x17e9cb96b (filename not found) [in UE4Editor-AirSim.dylib]
msr::airlib::MavLinkDroneController::start() Address = 0x17e9cb937 (filename not found) [in UE4Editor-AirSim.dylib]
MultiRotorConnector::beginPlay() Address = 0x17eb077fd [/Users/shatadalmishra/Documents/Unreal Projects/UAVProject/Plugins/AirSim/Source/MultiRotorConnector.cpp, line 68] [in UE4Editor-AirSim.dylib]
ASimModeWorldBase::createWorld() Address = 0x17eb13389 [/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iterator, line 1190] [in UE4Editor-AirSim.dylib]
ASimModeWorldMultiRotor::BeginPlay() Address = 0x17eb1c37e [/Users/shatadalmishra/Documents/Unreal Projects/UAVProject/Plugins/AirSim/Source/SimMode/SimModeWorldBase.cpp, line 25] [in UE4Editor-AirSim.dylib]
AActor::DispatchBeginPlay() Address = 0x10ee955f8 (filename not found) [in UE4Editor-Engine.dylib]
AActor::PostActorConstruction() Address = 0x10ee94249 (filename not found) [in UE4Editor-Engine.dylib]
AActor::FinishSpawning(FTransform const&, bool, FComponentInstanceDataCache const*) Address = 0x10ee905a5 (filename not found) [in UE4Editor-Engine.dylib]
AActor::PostSpawnInitialize(FTransform const&, AActor*, APawn*, bool, bool, bool) Address = 0x10ee8d78c (filename not found) [in UE4Editor-Engine.dylib]
UWorld::SpawnActor(UClass*, FTransform const*, FActorSpawnParameters const&) Address = 0x10fa7961c (filename not found) [in UE4Editor-Engine.dylib]
UWorld::SpawnActor(UClass*, FVector const*, FRotator const*, FActorSpawnParameters const&) Address = 0x10fa7a0bb (filename not found) [in UE4Editor-Engine.dylib]
ASimHUD::BeginPlay() Address = 0x17eb0e308 [Runtime/Engine/Classes/Engine/World.h, line 2695] [in UE4Editor-AirSim.dylib]
AActor::DispatchBeginPlay() Address = 0x10ee955f8 (filename not found) [in UE4Editor-Engine.dylib]
AWorldSettings::NotifyBeginPlay() Address = 0x11079978a (filename not found) [in UE4Editor-Engine.dylib]
AGameStateBase::HandleBeginPlay() Address = 0x10f85b851 (filename not found) [in UE4Editor-Engine.dylib]
UWorld::BeginPlay() Address = 0x11077956e (filename not found) [in UE4Editor-Engine.dylib]
UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&) Address = 0x10f812c4a (filename not found) [in UE4Editor-Engine.dylib]
UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float) Address = 0x115bd1c86 (filename not found) [in UE4Editor-UnrealEd.dylib]
UEditorEngine::PlayInEditor(UWorld*, bool) Address = 0x115bc6cc8 (filename not found) [in UE4Editor-UnrealEd.dylib]
UEditorEngine::StartQueuedPlayMapRequest() Address = 0x115bae6b4 (filename not found) [in UE4Editor-UnrealEd.dylib]
UEditorEngine::Tick(float, bool) Address = 0x1153d3a08 (filename not found) [in UE4Editor-UnrealEd.dylib]
UUnrealEdEngine::Tick(float, bool) Address = 0x11603979c (filename not found) [in UE4Editor-UnrealEd.dylib]
FEngineLoop::Tick() Address = 0x10d937866 (filename not found) [in UE4Editor]
GuardedMain(wchar_t const*) Address = 0x10d93ef92 (filename not found) [in UE4Editor]
-[UE4AppDelegate runGameThread:] Address = 0x10d94c6ba (filename not found) [in UE4Editor]
-[FCocoaGameThread main] Address = 0x10db7a506 (filename not found) [in UE4Editor-Core.dylib]
Unknown() Address = 0x7fffa3cf6b3d (filename not found) [in Foundation]
_pthread_body Address = 0x7fffb7c6693b (filename not found) [in libsystem_pthread.dylib]
_pthread_body Address = 0x7fffb7c66887 (filename not found) [in libsystem_pthread.dylib]
thread_start Address = 0x7fffb7c6608d (filename not found) [in libsystem_pthread.dylib]

I am on a Macbook Pro . Any help is appreciated.

Thanks And Regards,
Shatadal Mishra

lovettchris, 2017-06-08:
Looks like createMavSerialConnection is throwing an exception, it does this for the following reasons:

```
        std::string port_name_auto = port_name;
        if (port_name_auto == "" || port_name_auto == "*") {
            port_name_auto = findPixhawk();
            if (port_name_auto == "") {
                throw std::domain_error("Could not find a connected PX4 flight controller");
            }
        }

        if (port_name_auto == "") {
            throw std::invalid_argument("SerialPort setting has an invalid value.");
        }

        if (baud_rate == 0) {
            throw std::invalid_argument("SerialBaudRate has an invalid value");
        }
```

So try specifying the correct SerialPort name in your ~/Documents/AirSim/settings.json file like this:

```
{
  "LocalHostIp": "127.0.0.1",
  "Pixhawk": {
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "/dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
```

smishraasu, 2017-06-09:
Thank You for the response. Now I was able to see a quadcopter in the screen. But now I am getting the following error. 

Vehicle controller cannot be started. please check your settings.json std::exception. 
[image]

lovettchris, 2017-06-09:
check that "/dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00" actually exists using "ls" command, your controller might have slightly different name.

smishraasu, 2017-06-09:
Thank you for the swift response chris. I have been able to connect to HIL mode. But my quadcopter doesn't take off. When I check in the qgroundcontrol the HUD seems to be all over the place. Even though the quadcopter is still, the HUD is moving rapidly. I have attached a screenshot. 

There seems to be some issue with the sensor data. 

[image]

lovettchris, 2017-06-09:
And the AirFrame is set to "HIL Quadrocopter X" ?

smishraasu, 2017-06-09:
My device name is "/dev/cu.usbmodem1".

Airframe is set to DJIF330 on qgc. I will do change that on qgroundcontrol and check if I am able to take off. 

Thanks Chris

smishraasu, 2017-06-09:
I have finally had a successful simulator flight on MacBook Pro. Thanks Chris for the help.

closed 2017-06-09 by smishraasu

lovettchris, 2017-06-09:
Awesome, thanks for reporting your final result.  I hear that MacBook Pro gets about 20 fps in a simple world, right?

smishraasu, 2017-06-09:
Honestly, I haven't tried a simple world yet. My next step is to create a simple world and try it. I will update the results here. I have just tried the basic no world scenario I believe. In that it is 60 fps, because I guess there is nothing to render.

Thanks.

lovettchris, 2017-06-09:
Cool, we provided one in the ~/AirSim/Unreal/Environments/Blocks folder.

smishraasu, 2017-06-09:
That makes my life a bit easier. I will try and report the fps.

smishraasu, 2017-06-09:
Hi Chris, while trying to integrate the environment in the Blocks folder, based on your youtube video, I am getting the following error in Unreal engine 

[image]

Could you let me know of the steps for seamlessly integrating environment in Blocks folder.

lovettchris, 2017-06-09:
I think you can safely ignore that error.

smishraasu, 2017-06-10:
Hi Chris,

In Block on Macbook Pro, the Fps is b/w 50-55. I think that is more than enough for development and testing. Would you be knowing of any environment which is a bit more complicated than Blocks and much less complicated than Landscapes ? I would be interested in trying those out to check the FPS.

Thanks.