microsoft/AirSim #267: Stabilizing camera images/Adding a gimbal

hngenc, 2017-06-08:
We're running tracking algorithms on images we get from AirSim's cameras. However, every time the drone stops or changes direction, the camera "wobbles", causing our tracking algorithm to fail.

Is there some way to stabilize the FPV camera so that it's roll and pitch remain constant? I'd like to have the same effect that I would get from attaching a gimbal to the drone.

lovettchris, 2017-06-08:
Implementing a proper camera gimbal would be awesome (something [like this](https://www.youtube.com/watch?v=uBh60jDsdPU)).  You might be able to just set the fpv camera's rotation using SetActorRotation to cancel the orientation of the drone, perhaps setting some limits so camera doesn't look inside the drone in extreme cases... see CameraDirector and UAirBlueprintLib::FollowActor for example of how we do that with the external chase camera.

hngenc, 2017-06-11:
@lovettchris Thanks! I managed to simulate a gimbal by doing as you said. For anyone else interested in doing the same, I added `#include "VehiclePawnBase.h"` to AirBlueprintLib.cpp and then added the following three lines to the end of UAirBlueprintLib::FollowActor:

`AActor * fpv_cam = ((AVehiclePawnBase*)followee)->getFpvCamera();
FRotator drone_orientation = ((AVehiclePawnBase*)followee)->getOrientation();
fpv_cam->SetActorRotation(FRotator(0, drone_orientation.Yaw, 0));`

lovettchris, 2017-06-12:
Hey congrats, that is cool to know it can be done in only 3 lines of code.  Thanks for posting the answer!

closed 2017-06-12 by lovettchris

Mohamed-Kassem, 2018-01-27:
@lovettchris After the modification in the repo, I changed added `#include "**VehiclePawnWrapper**.h"`
and these three lines to the end of `UAirBlueprintLib::FollowActor`

`AActor * fpv_cam = ((**VehiclePawnWrapper***)followee)->**getCamera**();
FRotator drone_orientation = ((**VehiclePawnWrapper***)followee)->getOrientation();
fpv_cam->SetActorRotation(FRotator(0, drone_orientation.Yaw, 0));`

and successfully recompiled AirSim, when I open the blocks Environment with the car it runs fine, after pressing `f` unreal crashes, and if I choose to run with the multirotor unreal also crashes, do you how could I solve this problem, or how to add the gimbal in the last commit in the repo?

Thanks a lot.