microsoft/AirSim #268: yaw right control trigger RTL(return to launch)

TomorrowIsAnOtherDay, 2017-06-09:
I'm using TGFG-tutaba Ratio, all the others control work on the simulator except the yaw right controll, which triggered RTL like this:
[image]

lovettchris, 2017-06-09:
it is very unlikely that controlling yaw with a stick would trigger RTL.  The RTL setup can be shown in QGroundControl, the reasons for RTL are listed under the Safety tab, these are my settings, notice for simulation I disable RC loss failsafe and data link loss failsafe, but I wouldn't do that on a real drone.

[image]

TomorrowIsAnOtherDay, 2017-06-09:
thanks for your reply.After resetting the flight control and calibrate the ratio and flight mode, It worked.

closed 2017-06-09 by TomorrowIsAnOtherDay

lovettchris, 2017-06-09:
congrats and welcome to the club :-)