microsoft/AirSim #271: How to record depth?

rmalav15, 2017-06-10:
I am sorry for the newbie question. I must be mission something obvious.

I have build the Airsim and running it with sitl simulator.
When I press the record button, it only store screenshots into my document/Airsim folder.
Neither depth nor segmentation images are stored. Please let me know the solution.

Mayankm96, 2017-06-10:
Hey! 

You could easily do that using the [python API](https://github.com/Microsoft/AirSim/blob/master/PythonClient/camera.py) or [C++ API](https://github.com/Microsoft/AirSim/blob/master/HelloDrone/main.cpp) by using the openCV function `inwrite`. 

Another way which is typically better is to change `EPIPCameraType::PIP_CAMERA_TYPE_SCENE` to the desired type,  (in this case `EPIPCameraType::PIP_CAMERA_TYPE_DEPTH)` in line 41 of the [RecordingThread.cpp](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/RecordingThread.cpp):

`             USceneCaptureComponent2D* capture = cam->getCaptureComponent(EPIPCameraType::PIP_CAMERA_TYPE_SCENE, true);
`
This changes the record button to record depth images instead of scene images. 

Hope that helps!

rmalav15, 2017-06-10:
Hey thanks for quick reply.
I saw that you have opened a issue earlier https://github.com/Microsoft/AirSim/issues/244.
I need to do exactly the same. can you share you python API with me that record both view and depth simultaneously? It will be a huge help.

Mayankm96, 2017-06-10:
Yeah well you can find the script [here](https://github.com/Mayankm96/ais-work/blob/master/airsim-python/camera.py). If you're trying to record the images synchronously you might face a bit of trouble. I am still thinking a way out to do this.

rmalav15, 2017-06-10:
Ok much thanks. So I need to run both the unreal editor environment and this script simultaneously. And it will record both things. And I am not much worried about synchronization. A small gap would work for me.

And Again much thanks.

Mayankm96, 2017-06-10:
Haha well the frame rate is not good and the little gap ends up being large. At least on my system.

sytelus, 2017-07-09:
There have been few contributions that have made FPS much smoother while recording images. I would suggest try out the new version!

closed 2017-07-09 by sytelus

sytelus, 2017-12-12:
The feature to record images from multiple cameras simultaneously is now available. You can, for example, record scene and depth images simultaneously. 

Please see the doc: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md#recording

dehuachen2017, 2018-03-08:
I could not find "EPIPCameraType::PIP_CAMERA_TYPE_SCENE" in RecordingThread.cpp (please see attached file). What can I do in order to be able to record depth and scene simultaneously?

[RecordingThread.txt](https://github.com/Microsoft/AirSim/files/1792378/RecordingThread.txt)