microsoft/AirSim #272: Fatal error crash on Ubuntu 16.04

kevindehecker, 2017-06-12:
I'm running Ubuntu 16.04 with UE4 release 4.15.3 and Airsim master (pulled again just now). I've create a new UE c++ project with no startercontect (just the one bottom plate) and installed the Airsim plugin into that. When I set the GameMode Override to AirsimGameMode it crashes with the error pasted below as soon as I hit the play button.

Running the same world with GameMode Override set to none works OK. 
A similar Windows based setup (same Pixhawk hardware and also UE 4.15.3, Airsim plugin is a few weeks older but I don't dare to pull :) ) also works OK with AirsimGamemode.

The error:
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
[2017.06.12-09.31.14:522][448]LogCrashTracker: 

[2017.06.12-09.31.14:523][448]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000002

[2017.06.12-09.31.14:523][448]LogLinux: Fatal error!

[Callstack]  00  0x00007fcca06cb8ff  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[Callstack]  01  0x00007fcca0585c95  FGenericPlatformStackWalk::StackWalkAndDump(char*, unsigned long, int, void*)
[Callstack]  02  0x00007fcca06836b1  FLinuxCrashContext::CaptureStackTrace()
[Callstack]  03  0x00007fcc956cc340  CommonLinuxCrashHandler(FGenericCrashContext const&)
[Callstack]  04  0x00007fcca06885b2  PlatformCrashHandler(int, siginfo_t*, void*)
[Callstack]  05  0x00007fcca64b1390  /lib/x86_64-linux-gnu/libpthread.so.0(+0x11390) [0x7fcca64b1390]
[Callstack]  06  0x00007fcca06177a8  FMallocBinned::Private::FreeInternal(FMallocBinned&, void*)
[Callstack]  07  0x00007fcc3441e61a  std::__1::basic_filebuf<char, std::__1::char_traits<char> >::~basic_filebuf()
[Callstack]  08  0x00007fcc34417679  msr::airlib::Settings::loadJSonFile(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >)
[Callstack]  09  0x00007fcc34405f31  ASimModeBase::initializeSettings()
[Callstack]  10  0x00007fcc34405c87  ASimModeBase::BeginPlay()
[Callstack]  11  0x00007fcc3440f8b2  ASimModeWorldMultiRotor::BeginPlay()
[Callstack]  12  0x00007fcc9be9a95c  AActor::DispatchBeginPlay()
[Callstack]  13  0x00007fcc9be99b87  AActor::PostActorConstruction()
[Callstack]  14  0x00007fcc9be975a3  AActor::FinishSpawning(FTransform const&, bool, FComponentInstanceDataCache const*)
[Callstack]  15  0x00007fcc9be9533a  AActor::PostSpawnInitialize(FTransform const&, AActor*, APawn*, bool, bool, bool)
[Callstack]  16  0x00007fcc9c75e5cf  UWorld::SpawnActor(UClass*, FTransform const*, FActorSpawnParameters const&)
[Callstack]  17  0x00007fcc9c75f2dc  UWorld::SpawnActor(UClass*, FVector const*, FRotator const*, FActorSpawnParameters const&)
[Callstack]  18  0x00007fcc34404b52  ASimHUD::BeginPlay()
[Callstack]  19  0x00007fcc9be9a95c  AActor::DispatchBeginPlay()
[Callstack]  20  0x00007fcc9d153210  AWorldSettings::NotifyBeginPlay()
[Callstack]  21  0x00007fcc9c5e199c  AGameStateBase::HandleBeginPlay()
[Callstack]  22  0x00007fcc9d13def9  UWorld::BeginPlay()
[Callstack]  23  0x00007fcc9c5a4bce  UGameInstance::StartPlayInEditorGameInstance(ULocalPlayer*, FGameInstancePIEParameters const&)
[Callstack]  24  0x00007fcc9716e00d  UEditorEngine::CreatePIEGameInstance(int, bool, bool, bool, bool, float)
[Callstack]  25  0x00007fcc97165ee4  UEditorEngine::PlayInEditor(UWorld*, bool)
[Callstack]  26  0x00007fcc9715626a  UEditorEngine::StartQueuedPlayMapRequest()
[Callstack]  27  0x00007fcc96b4991f  UEditorEngine::Tick(float, bool)
[Callstack]  28  0x00007fcc9751c039  UUnrealEdEngine::Tick(float, bool)
[Callstack]  29  0x0000000000423de0  FEngineLoop::Tick() [Runtime/Core/Public/Stats/Stats2.h, line 1686]
[Callstack]  30  0x000000000042b923  GuardedMain(wchar_t const*) [/mnt/Bla/UnrealEngine-4.15.3-release/Engine/Source/Runtime/Launch/Private/Launch.cpp, line 62]
[Callstack]  31  0x00007fcc956cd0ca  CommonLinuxMain(int, char**, int (*)(wchar_t const*))
[Callstack]  32  0x00007fcc94a45830  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0) [0x7fcc94a45830]
[Callstack]  33  0x0000000000415e29  /mnt/Bla/UnrealEngine-4.15.3-release/Engine/Binaries/Linux/UE4Editor(_start+0x29) [0x415e29]

[2017.06.12-09.31.14:549][448]LogExit: Executing StaticShutdownAfterError
Malloc Size=263811 LargeMemoryPoolOffset=460483 
Using binned.
LogPlatformFile: Not using cached read wrapper
LogInit:Display: RandInit(-84552496) SRandInit(-84552496).
LogTaskGraph: Started task graph with 4 named threads and 7 total threads with 1 sets of task threads.
LogInit: Build: ++UE4+Release-4.15-CL-0
LogInit: Engine Version: 4.15.3-0+++UE4+Release-4.15
LogInit: Compatible Engine Version: 4.15.0-0+++UE4+Release-4.15
LogInit: Net CL: 0
LogInit: Compiled (64-bit): Jun  9 2017 17:02:31
LogInit: Compiled with Clang: 3.9.1 (branches/release_39)
LogInit: Build Configuration: Shipping
LogInit: Branch Name: ++UE4+Release-4.15
LogInit: Command line:  -Abslog=/home/houjebek/Documents/Unreal Projects/MyProject/Saved/Logs/MyProject-CRC.log /home/houjebek/Documents/Unreal Projects/MyProject/Saved/Crashes/crashinfo-MyProject-pid-32318-21A872A70E060B0C001F0225D9B37AE1/
LogInit: Base directory: /mnt/Bla/UnrealEngine-4.15.3-release/Engine/Binaries/Linux/
LogInit: Installed Engine Build: 1
LogInit: Presizing for max 100000 objects, including 0 objects not considered by GC, pre-allocating 0 bytes for permanent pool.
[... 12 lines cut ...]
[2017.06.12-09.31.14:718][  0]LogInit:  - Cache line size: 64
[2017.06.12-09.31.14:718][  0]LogInit:  - Memory allocator used: binned
[2017.06.12-09.31.14:718][  0]LogLinux: Skipped benchmarking clocks because the engine is running in a standalone program mode - CLOCK_REALTIME will be used.
[2017.06.12-09.31.14:718][  0]LogInit: Linux-specific commandline switches:
[2017.06.12-09.31.14:718][  0]LogInit:  -nodwarf (currently OFF): suppress parsing of DWARF debug info (callstacks will be generated faster, but won't have line numbers)
[2017.06.12-09.31.14:718][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[2017.06.12-09.31.14:718][  0]LogInit:  -jemalloc - use jemalloc for all memory allocation
[2017.06.12-09.31.14:718][  0]LogInit:  -binnedmalloc - use binned malloc  for all memory allocation
[2017.06.12-09.31.14:718][  0]LogInit:  -httpproxy=ADDRESS:PORT - redirects HTTP requests to a proxy (only supported if compiled with libcurl)
[2017.06.12-09.31.14:718][  0]LogInit:  -reuseconn - allow libcurl to reuse HTTP connections (only matters if compiled with libcurl)
[2017.06.12-09.31.14:718][  0]LogInit:  -virtmemkb=NUMBER - sets process virtual memory (address space) limit (overrides VirtualMemoryLimitInKB value from .ini)
[2017.06.12-09.31.14:718][  0]LogInit:  - Physical RAM available (not considering process quota): 32 GB (31898 MB, 32663640 KB, 33447567360 bytes)
[2017.06.12-09.31.14:718][  0]LogUObjectArray: 146 objects as part of root set at end of initial load.
[2017.06.12-09.31.14:718][  0]LogUObjectAllocator: 24096 out of 0 bytes used by permanent object pool.
[2017.06.12-09.31.14:718][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2017.06.12-09.31.14:718][  0]LogTextLocalizationManager: No translations for ('') exist, falling back to 'en' for localization and internationalization data.
[2017.06.12-09.31.14:720][  0]CrashReportClientLog: CrashReportClientVersion=1.0
[2017.06.12-09.31.14:720][  0]CrashReportClientLog: CrashReportReceiver disabled
[2017.06.12-09.31.14:720][  0]CrashReportClientLog: DataRouterUrl: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data
[2017.06.12-09.31.14:720][  0]LogExit: Preparing to exit.
[2017.06.12-09.31.14:720][  0]LogObj: Freed 0b from 0 cluster array pools.
[2017.06.12-09.31.14:720][  0]LogExit: Object subsystem successfully closed.
[2017.06.12-09.31.14:720][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.06.12-09.31.14:721][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.

kevindehecker, 2017-06-12:
FYI, I did find issue #112, but it was closed loosely mentioning that some changes where committed to the project. Apparently the problem re-appeared or was not solved completely.

iamvoid13, 2017-06-12:
I'm also facing the same issue. Whenever I try to press Play Button the Unreal Studio Crashes with Segmentation Fault. I've been stuck on this issue for quite some days. I'm running Ubuntu 16.04. Any help will be appreciated.

Smoolak, 2017-06-15:
I also have the same issue. As described before when the game mode is set on AirsimGameMode the UE4Editor crash as soon as you hit play.

The distribution I use is Arch Linux. My first try was to compile UE4 4.15.0 and AirSim with clang 3.5. I then created a project and added AirSim to the project folder and the uproject file as shown in the set up video, then I changed the game mode to AirsimGameMode, clicked play and boom I got my first crash. For my second try I did compile AirSim with clang 4.0.0 while keeping UE4 4.15.0, the same thing happened. For my third try I compiled UE4 4.15.3 and AirSim with clang 4.0.0 and I got the same crash and the same error.

Here's what I did to get to this point:
1. I downloaded a snapshot of the unreal-engine package from AUR for the version of UE4 I wanted to try (https://aur.archlinux.org/cgit/aur.git/log/?h=unreal-engine). Its easier this way since the author of the package already added some patch for the compilation to work on Arch Linux.
2. I extracted the archive and runned `makepkg -sri`. At this point I waited for the compilation and the installation to finish.
3 I lunched the UE4Editor and created an empty C++ project.
4. I git cloned the AirSim repository.
5. I successfully compiled AirSim with the build.sh script.
6. I added the Plugins folder from AirSim to my UE4 project.
7. I edited the uproject file in my project as indicated here: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md
8. I executed `./GenerateProjectFiles.sh -project=<path to project>/<project name>.uproject -game -engine`
9. I then launched my project with the UE4Editor like this: `UE4Editor <path to project>/<project name>.uproject`
10. I changed the game mode to AirsimGameMode.
11. I clicked play, then the editor crashed with this error:

[2017.06.15-16.25.53:199][260]LogTemp: Repeting last play command: Selected Viewport
[2017.06.15-16.25.53:204][260]LogBlueprintUserMessages: Early PlayInEditor Detection: Level '/Game/StarterContent/Maps/Minimal_Default.Minimal_Default:PersistentLevel' has LevelScriptBlueprint '/Game/StarterContent/Maps/Minimal_Default.Minimal_Default:PersistentLevel.Minimal_Default' with GeneratedClass '/Game/StarterContent/Maps/Minimal_Default.Minimal_Default_C' with ClassGeneratedBy '/Game/StarterContent/Maps/Minimal_Default.Minimal_Default:PersistentLevel.Minimal_Default'
[2017.06.15-16.25.53:205][260]BlueprintLog: New page: Pre-Play auto-recompile
[2017.06.15-16.25.53:205][260]LogPlayLevel: [PlayLevel] Compiling BP_SimHUDWidget before play...
[2017.06.15-16.25.53:320][260]LogPlayLevel: PlayLevel: Blueprint regeneration took 115 ms (1 blueprints)
[2017.06.15-16.25.53:324][260]PIE: New page: PIE session: Minimal_Default (Jun 15, 2017, 10:25:53 AM)
[2017.06.15-16.25.53:325][260]LogPlayLevel: Creating play world package: /Game/StarterContent/Maps/UEDPIE_0_Minimal_Default
[2017.06.15-16.25.53:339][260]LogPlayLevel: PIE: StaticDuplicateObject took: (0.013668s)
[2017.06.15-16.25.53:344][260]LogAIModule: Creating AISystem for world Minimal_Default
[2017.06.15-16.25.53:344][260]LogPlayLevel: PIE: World Init took: (0.005524s)
[2017.06.15-16.25.53:345][260]LogPlayLevel: PIE: Created PIE world by copying editor world from /Game/StarterContent/Maps/Minimal_Default.Minimal_Default to /Game/StarterContent/Maps/UEDPIE_0_Minimal_Default.Minimal_Default (0.019559s)
[2017.06.15-16.25.53:364][260]LogALAudio: ALAudio device opened : OpenAL Soft
[2017.06.15-16.25.53:365][260]LogALAudio: AL_VENDOR      : OpenAL Community
[2017.06.15-16.25.53:365][260]LogALAudio: AL_RENDERER    : OpenAL Soft
[2017.06.15-16.25.53:365][260]LogALAudio: AL_VERSION     : 1.1 ALSOFT 1.15.1
[2017.06.15-16.25.53:365][260]LogALAudio: AL_EXTENSIONS  : AL_EXT_ALAW AL_EXT_DOUBLE AL_EXT_EXPONENT_DISTANCE AL_EXT_FLOAT32 AL_EXT_IMA4 AL_EXT_LINEAR_DISTANCE AL_EXT_MCFORMATS AL_EXT_MULAW AL_EXT_MULAW_MCFORMATS AL_EXT_OFFSET AL_EXT_source_distance_model AL_LOKI_quadriphonic AL_SOFT_buffer_samples AL_SOFT_buffer_sub_data AL_SOFTX_deferred_updates AL_SOFT_direct_channels AL_SOFT_loop_points AL_SOFT_source_latency
[2017.06.15-16.25.53:365][260]LogInit: FAudioDevice initialized.
[2017.06.15-16.25.53:587][260]LogAudioDerivedData:Display: Building compressed audio format OGG wave StartPlayInEditor...
[2017.06.15-16.25.53:636][260]LogLoad: Game class is 'AirSimGameMode'
[2017.06.15-16.25.53:641][260]LogWorld: Bringing World /Game/StarterContent/Maps/UEDPIE_0_Minimal_Default.Minimal_Default up for play (max tick rate 0) at 2017.06.15-13.25.53
[2017.06.15-16.25.53:642][260]LogWorld: Bringing up level for play took: 0.002048
[2017.06.15-16.25.53:761][260]LogContentBrowser: Native class hierarchy updated for 'MovieSceneCapture' in 0.0044 seconds. Added 11 classes and 0 folders.
[2017.06.15-16.25.53:764][260]Cmd: r.CustomDepth 3
[2017.06.15-16.25.53:769][260]r.CustomDepth = "3"
Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
CommonLinuxCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=196655 
[2017.06.15-16.25.53:825][260]LogCrashTracker: 

[2017.06.15-16.25.53:825][260]LogLinux: === Critical error: ===
Unhandled Exception: SIGSEGV: invalid attempt to read memory at address 0x0000000000000002

[2017.06.15-16.25.53:826][260]LogLinux: Fatal error!

[Callstack]  00  0x00007f927a3a179f  FLinuxPlatformStackWalk::CaptureStackBackTrace(unsigned long long*, unsigned int, void*)
[... cut ...]
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
Found candidate GCC installation: /usr/bin/../lib/gcc/x86_64-pc-linux-gnu/7.1.1
Found candidate GCC installation: /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/7.1.1
Found candidate GCC installation: /usr/lib/gcc/x86_64-pc-linux-gnu/7.1.1
Found candidate GCC installation: /usr/lib64/gcc/x86_64-pc-linux-gnu/7.1.1
Selected GCC installation: /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/7.1.1
Candidate multilib: .;@m64
Selected multilib: .;@m64

jeffdelmerico, 2017-06-23:
I am also experiencing the same crash with the same setup (16.04, UE 4.15.3, empty C++ project) and the master branch as of today (June 23).  

The discussions for this issue and #176 seem to have stopped about a week ago...has anyone had any new insight or progress since then?  Or does anyone know of an older commit that works with Ubuntu 16.04 and UE 4.15.3?

abhay-agarwal, 2017-06-28:
My Linux build uses Ubuntu 16.04 and UE 4.15.3

This error is more of a class of errors than a single one. This can be caused by a malformed Json file or an incomplete one. Try using your Json file on windows to see if that works. Another approach is to try running the packager for Linux and try to package the build. Sometimes that process reveals the error in question.

Building and packaging on Linux is currently not officially supported, however. The build that worked for me has a patch that removes all references to the “record_file”. Then Unreal Engine needs to be patched to add AirSim’s compiler flags. We should have a complete, working setup of this in a few weeks time. In the meantime, try to recreate Kerry Snyder’s docker but with your branch of AirSim.

sytelus, 2017-07-27:
We been busy fixing various issues in Linux build. Now we have new code out that should build and run on Ubuntu 16 with few known issues. Please see [docs](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)) for instructions.

closed 2017-07-27 by sytelus