microsoft/AirSim #273: camera does not follow the drone anymore

gregoryps1989, 2017-06-12:
At first everything was fine and the camera just followed my drone, but then I changed something unintentionally and now the camera as well as the drone are controlled by my controller at the same time and I do not know how to fix it. 
I tried things like search 'camera' and chose their options like 'snap view to object', but this did not change the problem.
Is there a way to reset the camera settings or tell the camera to automatically follow the drone?

lovettchris, 2017-06-12:
Did you accidentally press '[' or ']' on your keyboard?

[image]

gregoryps1989, 2017-06-17:
Non of the Keys to adjust the camera worked, so I guess this was not the problem.
I worked it out the following way, maybe there is more information in it:

I inserted a drone, a camera manager and a camera as you mention in your youtube setup video at the end. The default setting is still wrong, pressing 'play' does not switch to the camera view. But double clicking the BP_PIPCamera in the upper right window (1), inserts the camera view in a small window (2), which can be pined(3) and then right-clicked 'play from here'. This switches to the desired view. 
I still do not know what exactly went wrong and how to set this view as default, but it works.
 
[image]

closed 2017-07-11 by sytelus

kylelindgren, 2017-08-01:
Hi @gregoryps1989, just making sure you haven't changed the 'Play' option to something other than 'Selected Viewport'. I changed mine once and screwed up the camera-vehicle buddy system.
[image]

EliteTUM, 2017-08-02:
I have the same Issue. The Viewport is not the issue. Or better to say: Even with "Selected Viewport" checked, the View cannot be switched to FPV or "fly with me" or "ground observer view" on a Germany Keyboard Layout. I guess this still is the same for spanish keyboard layouts.

Switching the layout to English (EN-US) in Windows also didn't help.