microsoft/AirSim #274: Estimator Altitude Jumps after Home Position is Acquired in HITL

deonblaauw, 2017-06-12:
After selecting ARM on QGC, the relative altitude jumps to some random, higher value. It typically jumps anywhere between 12m and 25m. Takeoff and flight in Altitude mode is also uncontrollable (but stable, so inner-loops on rate are probably good), but then the interesting part happens. 

After a while, things would settle down and the vehicle would stabilize.  

I also noticed "jumps" in position on the map in QGC, so I suspected an issue with the GPS data being sent to the PixHawk, but upon further inspection I saw this (z-axis accelerometer):

[image]

The above image is only when "on ground".

When flying, the Z-Axis accelerometer looks like this:

[image]

I strongly suspect that, when the vehicle is "on ground", the ground dynamics model is adding acceleration noise to the Z-axis, which messes up the EKF.

Anyone else seen something similar?

I'm running on Windows 10, using QGC v3.1.3, PX4 release v1.6.1 and Unreal v4.15.3.

lovettchris, 2017-06-12:
Yes, I believe there is an bug here, I've seen the same thing.

deonblaauw, 2017-06-13:
I don't know the implementation / architecture that well yet, any idea where to start looking for it? I would think that it should be solved in the ground model (if that is even the name), I think X-Plane had something similar at one stage and it was patched but I'm not too clear on the details.

lovettchris, 2017-06-13:
Something to do with collisions, I think it is acting like it is hitting the ground, the ground pushes back which cancels gravity, so zacc jumps to 0, then gravity is calculated again which makes zacc go back to -9.8, then it hits the ground again, the ground pushes back, and over and over... I think the code instead should understand it is resting on the ground and therefore not to do any of the collision stuff.

sytelus, 2017-08-14:
I've made changes to baro model and parameters to resolve this issue. Please let us know if you still see this issue.

closed 2017-08-14 by sytelus

realdealneil, 2017-11-10:
I think I'm seeing this issue.  Attitude and altitude do very strange things when sitting on the ground.  As soon as I take off in stabilize mode, things look good again.  I am running: 

AirSim from Sep 5 (git commitish a5b79a06e0663102fefdcb5ff7f700a5353fe1be)
PX4 firmware: 1.6.5dev (updated today, 11/10/2017)

This is making it challenging to do offboard control testing.  Where is the fix committed?  What were the changes that you made so I can see if they've been applied in my version?

realdealneil, 2017-11-10:
I'm certain that I'm seeing this issue.  The z accelerometer jumps between -9.81 and 0 rapidly when I'm on the ground, but as soon as I take off, it smooths out like in the picture that @deonblaauw posted above.  The solution mentioned above (tuning baro model and parameters) is not the right solution.  We need to do something like @lovettchris proposed: detect that we're on the ground, disable collision detection, and set gravity vector to -9.81, and then upon taking off, switch out of "ground" mode.  Has anyone tried looking at this?  If you give me a clue on where to start, I can take a look.