microsoft/AirSim #275: Scene image and Depth image not synchronized?

ShukuiZhang, 2017-06-13:
I obtain the scene image and the depth image using python API, with the following code (modified from camera.py):
`from PythonClient import *
import cv2
import time
import sys
import os
import csv
client = AirSimClient('127.0.0.1')

help = False

# fontFace = cv2.FONT_HERSHEY_SIMPLEX
# fontScale = 0.5
# thickness = 2
# textSize, baseline = cv2.getTextSize("FPS", fontFace, fontScale, thickness)
# print (textSize)
# textOrg = (10, 10 + textSize[1])
frameCount = 0
startTime=time.clock()
fps = 0

DATA_PATH = 'C:\\Users\\shzhang\Documents\AirSim\Images'

while True:
    # because this method returns std::vector<uint8>, msgpack decides to encode it as a string unfortunately.
    result2 = client.getImageForCamera(0, AirSimImageType.Depth)
    result1 = client.getImageForCamera(0, AirSimImageType.Scene)
    
    if (result1 == "\0" or result2 == "\0"):
        if (not help):
            help = True
            print("Please press '1' in the AirSim view to enable the Depth camera view")
    else:
        rawImage = np.fromstring(result1, np.int8)
        png = cv2.imdecode(rawImage, cv2.IMREAD_UNCHANGED)
        
        depthImage = np.fromstring(result2, np.int8)
        depthpng = cv2.imdecode(depthImage, cv2.IMREAD_UNCHANGED)
        
        cv2.imshow("Depth", png)
        cv2.imshow("Raw", depthpng)
        cv2.imwrite(os.path.join(DATA_PATH ,"img_" + str(frameCount) + ".png"), png)
        cv2.imwrite(os.path.join(DATA_PATH, "depth_" + str(frameCount) + ".png"), depthpng)`

However, the recorded scene image and the depth image seems not synchronized. As shown in the following: 
[image]
[image]
Is there any solution to reduce the drift besides slowing down the drone? I mean, is there a way to improve synchronization performance by tuning the code? Thanks.

lovettchris, 2017-06-13:
Right they are rendered when you ask for them, which means they will not be perfectly synchronized, Sai Vemprala wrote a handy script that captures the images via screen capture so they are in sync. See his post on [issue#244](https://github.com/Microsoft/AirSim/issues/244).

ShukuiZhang, 2017-06-20:
Thanks. I will close this one and switch to that thread.

closed 2017-06-20 by ShukuiZhang