microsoft/AirSim #276: How to reset the uav location

TomorrowIsAnOtherDay, 2017-06-13:
we're working on  reinforcement learning algorithm while using Airsim as our simulator. We want to reset the uav location in home point while uav hit something(collision count plus one).Is there an api for our work?

lovettchris, 2017-06-14:
There is currently no DroneControlBase API to "teleport" the drone back to a home point, but it would be easy to add.  I would not recommend using the PX4 stack for reinforcement learning because it takes a while for the attitude estimate to settle down after a crash (See [debugging attitude estimation](https://www.youtube.com/watch?v=d_FyjKDWQfc&feature=youtu.be)).  

But if you must use PX4 the most certain way to get reliable results is to reboot the PX4, (restart SITL), but with SITL you could easily implement a mavlink "reset" command that tells SITL to stop and start the estimators (attitude/position).  This should be sufficient to flush any bad state and get drone back into clean into reproducible state.

sytelus, 2017-07-09:
Now we have API for this. You can use `simSetPosition(Vector3r(0,0,0))`.

closed 2017-07-09 by sytelus

TomorrowIsAnOtherDay, 2017-07-25:
I know little about the mavlink, Is there an easy way to reboot the PX4(restart the SITL)?

lovettchris, 2017-08-16:
rebooting PX4 can be done using the MavLinkCom class named MavLinkSerialControl, (see NshCommand in MavLinkTest for an example).  You can use this protocol to send NuttX command line commands, and one of those commands is named "reboot".

yousefab95, 2018-04-03:
Is simSetPosition(Vector3r(0,0,0)) still a part of the API? I'm trying to set the starting location of the drone by passing a set of coordinates.

walid19, 2018-12-27:
I am having the same problem, I need to use px4 for my work, I didn't get it how to reset the SITL programmatically!

clovett, 2018-12-28:
yousefab95, I use this code to teleport the drone to different locations on the map:
```python
import setup_path 
import airsim
import time

client = airsim.MultirotorClient()
client.confirmConnection()

pose = client.simGetVehiclePose()

# teleport the drone + 10 meters in x-direction
pose.position.x_val += 10

client.simSetVehiclePose(pose, True, "PX4")

time.sleep(2)

# teleport the drone back
pose.position.x_val -= 10

client.simSetVehiclePose(pose, True, "PX4")
```