microsoft/AirSim #277: block mavlink messages from sensors library

limzhanfeng, 2017-06-14:
Hi, I am trying send my own messages and would like to stop the current sensors library form sending out Mavlink message to the connection and updating the drone controller. How do I block these messages?

lovettchris, 2017-06-14:
See [intercepting mavlink](https://github.com/Microsoft/AirSim/wiki/Intercepting-MavLink-messages).

closed 2017-06-15 by limzhanfeng

nickylimjj, 2017-06-15:
Hi, I have just read [intercepting mavlink](https://github.com/Microsoft/AirSim/wiki/Intercepting-MavLink-messages) and tried to run mavlink test and run airsim after modifying `settings.json` as specified. However, i get the following error 

> not receiving any messages from HIL, please restart your HIL node and try again

Any clues what it may be?