microsoft/AirSim #278: Unable to connect Python Client with the AirSim server

avinashmaheshjoshi, 2017-06-14:
Hey,
I have got the AirSim plugin to work with Unreal on Windows 10. I had to comment

saihv, 2017-06-14:
I don't think the Python scripts would work if you set RpcEnabled to false, considering that's the main communication framework. Why did you have to set it to false?

avinashmaheshjoshi, 2017-06-14:
I had to comment the msbuild in debug mode as the build kept failing otherwise. Attached is the file with the console output.
[AirSim_Output.txt](https://github.com/Microsoft/AirSim/files/1075197/AirSim_Output.txt)

The reason for changing RpcEnabled to False was because the Unreal Engine kept crashing on running the drone. Screenshot below. I was trying a hit and trial approach to get it to work and the unreal engine stopped crashing when I set RpcEnabled to False.
[image]

Appreciate you trying to help me out. Please let me know if you need more information.

fshamshirdar, 2017-06-14:
Follow #129 in case you're using VS2017.
I have the same problem about crashing of the unreal engine, after some investigation I figured out that the RpcLib is the main issue in my case, that is why it works fine if you set RpcEnabled to false (by which python client won't be able to connect), Follow #176 for further information.

ethanyqc, 2017-06-16:
RpcLib should be set true, check if your in Debug mode rather than developer mode., As for fail to build.cmd, did you set your environmental variable for eigen and Cmake? You shouldn't need to comment out any part of the build

avinashmaheshjoshi, 2017-06-16:
I am using VS 2015 Community Edition. I'll do a clean re-install and try again.

I set environment variable for eigen to gitfolder\AirSim\eigen

I did not set anything for cmake manually as when I installed it asked for all users. Can you please tell me which variable do I need to verify is set or not?

Also, I am not sure if it's because of the Cmake variable but not only do I need to comment that line, I need to also add the below thing to the build statement of it to work. Or it cannot find the C++ compiler.
`
msbuild /p:Platform=x64 /p:Configuration=Release AirSim.sln /p:VisualStudioVersion=15.0`

avinashmaheshjoshi, 2017-06-16:
I uninstalled VS 2015 and all the C++ distributions. Installed it again using the ISO choosing the right options.

When I ran the build.cmd, it downloaded CMAKE again and installed it. So I guess it couldn't find my CMake installation.

Then it gave the following error. Does this mean that my Cmake variable is not set appropriately?

Link:
  C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\amd64\link.exe /ERRORREPORT:QUEUE /OUT:"C:\Users\Avinash\D
  esktop\AirSim\AirSim\DroneShell\build\x64\Debug\DroneShell.exe" /INCREMENTAL /NOLOGO /LIBPATH:C:\Users\Avinash\Deskto
  p\AirSim\AirSim\DroneShell\\..\AirLib\deps\MavLinkCom\lib\x64\Debug /LIBPATH:C:\Users\Avinash\Desktop\AirSim\AirSim\D
  roneShell\\..\AirLib\deps\rpclib\lib\x64\Debug /LIBPATH:C:\Users\Avinash\Desktop\AirSim\AirSim\DroneShell\\..\AirLib\
  lib\x64\Debug rpc.lib kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib
  oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /MANIFEST /MANIFESTUAC:"level='asInvoker' uiAccess='false'" /manifest:e
  mbed /DEBUG /PDB:"C:\Users\Avinash\Desktop\AirSim\AirSim\DroneShell\build\x64\Debug\DroneShell.pdb" /SUBSYSTEM:CONSOL
  E /TLBID:1 /DYNAMICBASE /NXCOMPAT /IMPLIB:"C:\Users\Avinash\Desktop\AirSim\AirSim\DroneShell\build\x64\Debug\DroneShe
  ll.lib" /MACHINE:X64 C:\Users\Avinash\Desktop\AirSim\AirSim\DroneShell\temp\DroneShell\x64\Debug\main.obj
  C:\Users\Avinash\Desktop\AirSim\AirSim\MavLinkCom\lib\x64\Debug\MavLinkCom.lib
  C:\Users\Avinash\Desktop\AirSim\AirSim\AirLib\lib\x64\Debug\AirLib.lib
AirLib.lib(client.obj) : error LNK2038: mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '0' doesn't match value '2
' in main.obj [C:\Users\Avinash\Desktop\AirSim\AirSim\DroneShell\DroneShell.vcxproj]

avinashmaheshjoshi, 2017-06-16:
Update: I had VS 2017 and VS 2015 both installed which probably was causing this issue. I uninstalled everything and just installed VS 2015. The build.cmd ran without any problems now.

But, when I copied the plugins folder in my unreal project and hit play, unreal engine crashes again. Same as above. When I hit debug, it breaks, below is the screenshot for the call stack and line number,

[image]

saihv, 2017-06-16:
Are you running in Development configuration or Debug Game Editor?

avinashmaheshjoshi, 2017-06-16:
When I open the MyProject.sln in VS 2015, the solution configuration was set to "Development Editor" configuration. I am not sure if I answered your question.

Apologies for my ignorance. If that was not the correct way of checking, please let me know where can I check.

saihv, 2017-06-16:
Oh yes, that's what I was referring to. Could you change that to Debug Game
editor and try?

avinashmaheshjoshi, 2017-06-17:
What do I do after changing it? 
I tried,

- Rebuild the solution, 
- Refresh the Visual Studio Project in UE4 editor
- Hit play, it crashed

avinashmaheshjoshi, 2017-06-20:
Hi,

I think I figured out what you meant. I changed it to "Debug Game editor" and built it from VS. It opened up the Unreal Engine Editor. Now the program does not crash with following settings.json,

{
  "LocalHostIp": "127.0.0.1",
  "Pixhawk": {
    "RemoteControlID": 0
  },
  "RpcEnabled": true
}

I am still unable to connect the python client to it though. Whenever I run takeoff.py, it prints the below message and pauses,

> Waiting for home GPS location to be set...

I tried running it line by line,

```
# connect to the AirSim simulator 
client = AirSimClient('127.0.0.1')
home = client.getHomePoint()
print home

: [nan, nan, nan]
```

Since home has nan, it is stuck in the next while loop.

```
while ((home[0] == 0 and home[1] == 0 and home[2] == 0) or
       math.isnan(home[0]) or  math.isnan(home[1]) or  math.isnan(home[2])):
    time.sleep(1)
    home = client.getHomePoint()
```

Am I missing a crucial step?

saihv, 2017-06-20:
Are you able to fly the drone manually, like with an RC?
Also could you change your settings.json to something like what's shown in this doc? 

https://github.com/Microsoft/AirSim/blob/master/docs/settings.md

avinashmaheshjoshi, 2017-06-21:
I don't have an RC. I was trying to set it up to be able to control it using the python client.

I have tried using the same settings.json with both setting UseSerial: true and false (Saw this in some other issue thread).

avinashmaheshjoshi, 2017-06-28:
Closing the issue as I was able to resolve this with the new settings.json pushed a few days back. Also I had to keep the SITL running before using python.

closed 2017-06-28 by avinashmaheshjoshi

sifengyang, 2017-07-23:
@avinashmaheshjoshi  I have same problem. when I run takeoff.py. It prints the message, "Waiting for home GPS location to be set...". How do you solve this problem?  thanks.

avinashmaheshjoshi, 2017-07-23:
Make sure SITL is running. The order is, start the project in Unreal, then run the SITL, then run your python script.

Follow this to setup SITL - https://github.com/Microsoft/AirSim/blob/ff5e664f49f4c99bb9fd71052e9f3305d78b4fba/docs/sitl.md