microsoft/AirSim #281: moveToPosition - Python

viniciusguigo, 2017-06-14:
Hello, all.

I was trying to implement "moveToPosition" in the Python API based on [this](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/controllers/DroneControllerBase.hpp).

This is what I did:

```
def moveToPosition(self, x, y, z, velocity, drivetrain, yaw_mode, lookahead, adaptive_lookahead):
            return self.client.call('moveToPosition',x, y, z, velocity, drivetrain, yaw_mode, lookahead, adaptive_lookahead)
```

But got this error:

```
File "/home/vinicius/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/vinicius/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "/home/vinicius/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 94, in send_message
    sock.send_message(message, callback)
  File "/home/vinicius/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 19, in send_message
    self._stream.write(self._packer.pack(message), callback=callback)
  File "msgpack/_packer.pyx", line 231, in msgpack._packer.Packer.pack (msgpack/_packer.cpp:3661)
  File "msgpack/_packer.pyx", line 233, in msgpack._packer.Packer.pack (msgpack/_packer.cpp:3503)
  File "msgpack/_packer.pyx", line 221, in msgpack._packer.Packer._pack (msgpack/_packer.cpp:3230)
  File "msgpack/_packer.pyx", line 221, in msgpack._packer.Packer._pack (msgpack/_packer.cpp:3230)
  File "msgpack/_packer.pyx", line 224, in msgpack._packer.Packer._pack (msgpack/_packer.cpp:3312)
  File "/home/vinicius/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 12, in <lambda>
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
AttributeError: 'YawMode' object has no attribute 'to_msgpack'
```
I would really appreciate if someone have a good idea to help me fix it.
Thank you!

lovettchris, 2017-06-14:
This [PythonClient.py API ](https://github.com/Microsoft/AirSim/blob/master/PythonClient/PythonClient.py)works and I've tested moveToPosition.  You'll see in there how YawMode has to be passed, namely, as a tuple.

viniciusguigo, 2017-06-14:
Thanks! I haven't pulled the new commit.

closed 2017-06-14 by viniciusguigo

zhangjunwang, 2019-04-16:
hi, I have the same problem with you. Could you tell me how to solve it? thanks

Traceback (most recent call last):
  File "pponew.py", line 325, in <module>
    client.setCarControls(car_controls)
  File "/home/zjw/.local/lib/python3.5/site-packages/airsim/client.py", line 305, in setCarControls
    self.client.call('setCarControls', controls, vehicle_name)
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 41, in call
    return self.send_request(method, args).get()
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/session.py", line 51, in send_request
    self._transport.send_message([message.REQUEST, msgid, method, args])
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 94, in send_message
    sock.send_message(message, callback)
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 19, in send_message
    self._stream.write(self._packer.pack(message), callback=callback)
  File "msgpack/_packer.pyx", line 284, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 290, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 287, in msgpack._packer.Packer.pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 263, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 234, in msgpack._packer.Packer._pack
  File "msgpack/_packer.pyx", line 277, in msgpack._packer.Packer._pack
  File "/home/zjw/.local/lib/python3.5/site-packages/msgpackrpc/transport/tcp.py", line 12, in <lambda>
    self._packer = msgpack.Packer(encoding=encodings[0], default=lambda x: x.to_msgpack())
AttributeError: 'numpy.float32' object has no attribute 'to_msgpack'

lovettchris, 2019-04-16:
Actually that is a different error:
```
AttributeError: 'numpy.float32' object has no attribute 'to_msgpack'
```
You might be able to fix this one by casting whatever values you are setting on the car_controls object to python float, so they do not remain as numpy.float32, something like this:
```
car_controls.steering = float(x)
```