microsoft/AirSim #282: questions about simulator

abdelrahman-t, 2017-06-14:
Hello, i have been using AirSim for the past few days, and it's AMAZING. I have a couple of questions though,  How should i go about extracting distance from the depth map images, Is there a scaling factor i could use to map 0-255 (int8) pixel values into a distance estimates (cm, m) range ?, how accurate would that estimate be ? , also how accurate is the gps measurement in meters ? (compared to uncertainty in gps measurements in real world) 

one last question 😄 , How do you estimate quadcopter velocity ? Do you integrate acceleration readings from the acc (optical flow, gps)?, How susceptible to drift would that estimate be  ?

thank you for the amazing project :)

ashkapoor, 2017-06-14:
Is this Abdo from Amazon?

abdelrahman-t, 2017-06-14:
no :)

lovettchris, 2017-06-14:
Glad you like AirSim.  Unreal recommends mapping 1 unit of measurement in an environment to 1 cm.  Not sure every environment abides by this, but AirSim uses this conversion internally and reports all x,y,z values through DroneControllerBase in meters and meters per second.   So as long as the unreal environment you are using stuck with this recommendation, the GPS values should be pretty accurate.

The depth map however is different - this is coming from Unreal SceneDepth texture.  There is [lots of discussion on Unreal forum ](https://answers.unrealengine.com/questions/332888/clarification-with-scenedepth-visualization.html) about how to calibrate those numbers.

For velocity, using PX4 based drones, the velocity is coming from PX4 position estimator via the MavLink message MAVLINK_MSG_ID_LOCAL_POSITION_NED.  Position estimation is a fusion of integrated accelerations and GPS sensor readings coming from AirSim.  We simulate "slower" GPS update rate of 200 ms, so dead reckoning for 200ms then update to GPS truth.  There is also some simulated noise in GPS just to make it realistic :-)   We have no optical flow simulation yet, but would love someone to add that.

abdelrahman-t, 2017-06-14:
thank you chris :)

Regards

closed 2017-06-18 by abdelrahman-t