microsoft/AirSim #283: executing mavlinktest as a proxy relay

nickylimjj, 2017-06-15:
Hi, I have just read [intercepting mavlink](https://github.com/Microsoft/AirSim/wiki/Intercepting-MavLink-messages) from #277 and tried to run mavlink test and airsim after modifying `settings.json` as specified. However, i get the following error 

> not receiving any messages from HIL, please restart your HIL node and try again

Any clues what it may be?

nickylimjj, 2017-06-19:
I have tried investigating further and when i run the command as shared `MavLinkTest.exe -serial -proxy:127.0.0.1:14560` and looked at my machine's networks using `netstat -abnp udp` i notice that port 14560 isn't in use, but mavlinktest.exe is is using `49722`. What should i do?

flyingsilverfin, 2017-07-19:
A month late but I noticed this can be solved by using -server instead of -proxy :)

closed 2017-08-14 by sytelus

enesvta, 2025-02-08:
> Bir ay geç oldu ama bunun -proxy yerine -server kullanılarak çözülebileceğini fark ettim :)

> A month late but I noticed this can be solved by using -server instead of -proxy :)

How did you?