microsoft/AirSim #284: How to add wind?

realflight1, 2017-06-15:
What would be the best / easiest way to apply wind to the simulated vehicle? Considering that I have the wind speed and direction.

realflight1, 2017-06-16:
Looking at `FastPhysicsEngine` it seems like one place to integrate wind would be in `getDragWrench` by modifying the actual vehicle velocity ([`vel_vertex`](https://github.com/Microsoft/AirSim/blob/593da156953a2514d7e02936f96f8752fb5b04af/AirLib/include/physics/FastPhysicsEngine.hpp#L206)). By modifying I mean adding the wind velocity vector in body frame to `vel_vertex`.

My knowledge of physics and aerodynamics is pretty basic, so I would appreciate any feedback. I used the following as a reference: [Does wind speed things up or accelerate things?](https://physics.stackexchange.com/questions/28827/does-wind-speed-things-up-or-accelerate-things)

sytelus, 2017-06-26:
You are almost there! Yes, you can just add wind vector to vehicle velocity to calculate final drag. However that's just one part of the story. Basically that simulates that if wind is towards vehicle then vehicle will feel resistance and reduce velocity. Other part of the wind effect is that wind can actually push vehicle. This is given by [very similar equation](http://www.engineeringtoolbox.com/wind-load-d_1775.html). Currently getDragWrench() would ignore the velocity vectors that do not have component towards vehicle (because it only computes drag). I would say you need another method getWindWrench() which computes wind pressure using formula from above link.

realdealneil, 2018-04-05:
Has this feature ever been implemented?  I need to simulate wind to make my controller more robust.

noNameForMe1994, 2018-06-28:
Is there anyway can I add the wind to the environment and get the real-time resistance or energy consumption of drone from airsim?

RVBldr, 2018-11-21:
Internal VSTF 7403

ardianumam, 2019-04-11:
Hi,
Is this already implemented yet?

Hiroskey, 2019-11-22:
Can I control the wind in real-time (i.e. dynamically)?

love-mj, 2020-04-14:
Has the effect of wind on UAV been realized？

rajat2004, 2020-07-21:
I've opened #2867 which adds Wind implementation according to the discussion here. Any testing or review, especially of the physics part would be great!

closed 2020-09-03 by madratman