microsoft/AirSim #285: Fixed wing model

lyhbuaa, 2017-06-16:
How can I add fixed wing models? Has anyone ever done such a thing?

lukaszdworako, 2017-06-21:
Currently it would require heavy modification, however the best place to start would be to change the rotor_z variable to be based around horizontal flight and find some kind of fixed wing model to import into the plugin, from what I've seen it may not be currently possible however you can still try. Just remember you must also make a new vehicle construction class in order to generate the motors properly and it would also be ideal to change rotor parameters to match that of the fixed wing model.

lyhbuaa, 2017-06-21:
ok, thank you very much!

lyhbuaa, 2017-06-21:
ok, thank you very much! @lukaszdworako

lukaszdworako, 2017-06-23:
@lyhbuaa if you are still trying to modify I accidentally referenced the wrong variable, the correct variable to control thrust direction is unit_z in MultiRotorParams.hpp. If you look through the easiest way to do this would be to create your own vehicle initializer. If more advice feel free to ask.

11tdisast3r, 2017-11-22:
I'm also interested to work on adding a fixed wing plane to Airsim

RVBldr, 2018-11-21:
Internal VSTF 7411.

jonyMarino, 2020-09-04:
treated in #2955

closed 2020-09-04 by jonyMarino