microsoft/AirSim #287: Unable to connect DroneServer to PX4 running on Remote machine

vaasudEva, 2017-06-16:
Hello, I've PX4 simulator running on a remote machine (Ubuntu).
I am interested to control it from HelloDrone running on localhost (Ubuntu).
I did change settings.json as below:
_{
  "LocalHostIp": "**remote-ip**",
  "RosFlight": {
  },
  "Pixhawk": {
    "LogViewerHostIp": "**remote-ip**",
    "LogViewerPort": 14388,
    "LogViewerSendPort": 14389,
    "Model": "Generic",
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "**remote-ip**",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "**remote-ip**",
    "SitlPort": 14556,
    "UdpIp": "**remote-ip**",
    "UdpPort": 14560,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  },
  "RpcEnabled": true
}_

But when I run DroneServer from command-line, I get 
`
$ ./cmake/output/bin/DroneServer 1
You are running in simulation mode.
terminating with uncaught exception of type std::runtime_error: UdpClientPort socket could not connect to remote host at 10.66.247.55:14560, error: 22
`
Could you please suggest right way of doing this ?

Thanks,
Prashanth

Smoolak, 2017-06-21:
I have the answer to this. For "LocalHostIp" you should use your local IP (its not the case in your example above). For example, I run the Unreal Engine with the simulation on a Windows PC and I run the PX4 controller and my HelloDrone on a Linux PC (I have to use that setup because of #272 ). The Windows PC have the IP 192.168.0.155 which is considered my local IP in this setup. The Linux PC have the IP 192.168.0.137 which is the remote IP.

Now, there is a problem with either the documentation at [settings.md](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) or the code at [DroneServer/main.cpp line 60](https://github.com/Microsoft/AirSim/blob/master/DroneServer/main.cpp#L60). Either "LocalHostIp" should be under "Pixhawk" or the code from the DroneServer should not read the "LocalHostIp" in the child. As a result, `connection_info.local_host_ip` defaults to 127.0.0.1 which provoke a connection error in [MavLinkCom/src/serial_com/UdpClientPort.cpp](https://github.com/Microsoft/AirSim/blob/6458eaf44aff5a802d9953457e06b0734f16d6b9/MavLinkCom/src/serial_com/UdpClientPort.cpp#L140) because `localaddr` and `remoteaddr` are not on the same interface. (@lovettchris you might whant to know about this)

To be clear, the quick fix for your problem is to use a config file that look like this:

{
"LocalHostIp": "**local-ip**",
"RosFlight": {
},
"Pixhawk": {
"LocalHostIp": "**local-ip**",
"LogViewerHostIp": "**remote-ip**",
"LogViewerPort": 14388,
"LogViewerSendPort": 14389,
"Model": "Generic",
"OffboardCompID": 1,
"OffboardSysID": 134,
"QgcHostIp": "**remote-ip**",
"QgcPort": 14550,
"SerialBaudRate": 115200,
"SerialPort": "*",
"SimCompID": 42,
"SimSysID": 142,
"SitlIp": "**remote-ip**",
"SitlPort": 14556,
"UdpIp": "**remote-ip**",
"UdpPort": 14560,
"UseSerial": false,
"VehicleCompID": 1,
"VehicleSysID": 135
},
"RpcEnabled": true
}

vaasudEva, 2017-06-29:
Dear  Smoolak, thank you so much for the reply.
But unfortunately, I get the same error even after above configuration.

`$ cmake/output/bin/DroneServer 1`
`You are running in simulation mode.`
`terminating with uncaught exception of type std::runtime_error: UdpClientPort socket could not connect to remote host at 10.66.247.55:14560, error: 22`

`Aborted (core dumped)`
What could be the issue ?

Smoolak, 2017-06-29:
@vaasudEva are your local IP on the same network as the remote one?

vaasudEva, 2017-06-29:
Yes both are on same network.

Smoolak, 2017-06-29:
Then my problem was unrelated to yours sorry about that.

vaasudEva, 2017-07-01:
It's fine.

sytelus, 2017-07-11:
You need to set LocalHostIp to IP address of server on both client and server machines. Make sure you can ping machine and access some other services just to make sure server is not behind firewall.

closed 2017-07-11 by sytelus