microsoft/AirSim #288: Custom Quads

eanmikale, 2017-06-17:
Chris, 
I know there was talk about creating the capability to utilize the DJI M100. We are interested in utilizing the M100, but are even more concerning the DJI Mavic Pro. Is this a possibility in the near future for either? Thank you in advance. 

EM

andre-nguyen, 2017-06-17:
Why not just add it in yourself by following the tutorial and using the official CAD?

https://github.com/Microsoft/AirSim/wiki/hexacopter
http://download.dji-innovations.com/downloads/dev/Matrice/M100_Airframe__CAD.STEP

closed 2017-06-17 by eanmikale

chaitanyantr, 2019-02-08:
hey @eanmikale ,

any luck on matric 100 in airsim??

eanmikale, 2019-02-08:
Yes, we just dropped it in as an asset. I know we are able to add our own custom hardware designs. However, we have inquiries concerning commercial hardware we are developing, which we would like to be an easy integration into the AirSim ecosystem. Is there anyone internally we can discuss our project with? Funny you mention the Matrix 100. Our projects seeks to replace it with an American made, truly open-source upgrade. Any assistance would be great. Support for the CrazyFlie swarms would be awesome as well. Thanks again. 

Regards,

Ean Mikale, J.D.

Chief Innovation Officer

Infinite 8 Institute, L3C

ean@infinite8institute.com

www.infinite8institute.com

U.S. Federal Contractor

---- On Fri, 08 Feb 2019 06:42:41 -0600 notifications@github.com wrote ----

hey @eanmikale ,

any luck on matric 100 in airsim??

demoniusrex, 2019-03-05:
I am working on a multi-rotor API implementation for DJI Onboard-SDK  that should support M100 with a few tweaks. All I have to test on is a N3 flight controller. 

You can check out the work in progress here [AirSim-Onboard](https://github.com/demoniusrex/AirSim-Onboard) (New multi-rotor API)[https://github.com/demoniusrex/AirSim-Onboard/blob/master/Rosie/linux/OnboardLib/include/vehicles/multirotor/firmwares/dji/OnboardMultirotorApi.hpp]