microsoft/AirSim #289: Async RPC Error

RodgerLuo, 2017-06-18:
I'm trying to implement a function in DroneShell to get collision information. To begin, I started by implementing a minor function to acquire Z position in DroneShell in order to study the structure of the codes. 

Based on the structure of the original codes, I added relevant codes in main.cpp of DroneShell, RpcLibClient.hpp, RpcLibClient.cpp, RpcLibServer.cpp, DroneControllerCancelable.hpp, DroneControllerBase.hpp, and DroneControllerBase.cpp. However, DroneShell returns "Async RPC Error: rpclib: server could not find function 'getZPos' with argument count 0". So I wonder if I miss to modify some other places? I can provide the codes I wrote if it's necessary. 

Thank you very much!

sytelus, 2017-06-26:
I just implemented getCollisionInfo() API which should now be available in Python.

https://github.com/Microsoft/AirSim/blob/master/AirLib/include/rpc/RpcLibClient.hpp#L46

RodgerLuo, 2017-07-05:
Thank you very much!

closed 2017-07-05 by RodgerLuo