microsoft/AirSim #291: Unable to move properly with DroneShell after taking off

ShukuiZhang, 2017-06-19:
I tried to make the drone fly with DroneShell, but after the drone taking off, any following command will make it go crazy. 
Win10
HIL

The following is what I did:
1. PX4 is purple blinking and I hit "play" in UE4, the drone shows up in simulator. 
2. I then open the DroneShell termial, it successfully find the GPS location. I then type "arm", still purple blinking. When I type "takeoff" afterwards, PX4 turns to be solid green and the drone takes off as expected.
3. I then tried a few commands: e.g. "movetoposition -x 1" and "rotate -duration 1" and many others. The drone will move a little bit and then gradually land. PX4 starts purple blinking and the screen shows "Failsafe mode enabled". I tried flying with a real RC and similar issues happened. Anyone has an idea? Thanks!

ShukuiZhang, 2017-06-20:
I just found the link: [https://github.com/Microsoft/AirSim/issues/201](url). I tried to set the "NAV_RCL_ACT=0" in QGroundControl->Vehicle Setup->Parameters by making it disabled. Is it a right way to do so? The problem still seems to exist....

viniciusguigo, 2017-06-26:
Hello,

The LED on the Pixhawk should not be purple at any point. When you spawn the quad it is supposed to be green (slow flashing). Please make sure there's  no warnings when you open your Pixhawk on QGroundControl.

This can also be caused by computer issues. Please check this link: https://github.com/Microsoft/AirSim/blob/master/docs/hard_drive.md

Thanks!

sytelus, 2017-06-29:
One other thing to try is also reset the parameters in QGC to default after you flush new firmware. Usually you don't have to but when I upgraded to new versions, sometime drone would behave erratically or just won't takeoff. I reset all params to default, re calibrate RC and set flight modes, things start working again.

closed 2017-07-09 by sytelus

zhuboshun2015, 2019-03-25:
hello!
 I have the same problem as you. I tried to move the drone after taking off with DroneShell(HIL) , but failed . Nothing happened after I entered the "MoveToZ -z -10" or "MoveToPosition -z -10 -x -100 -y -100".
Another thing is when I use DroneShell (HIL) to run the drone, it's position didn't change even after taking off ,but when I use RC to run the drone , it's position can reflect the true .
Anyone knows why?