microsoft/AirSim #293: Want to customize rotor directions

limzhanfeng, 2017-06-20:
Hello I would like to customize my own vehicle. I find the parameters for the dimensions and other things in rotors, multirotors and px4multirotor, but can't find where the normal directions of the rotors are set. How should I go about changing these settings?

lukaszdworako, 2017-06-21:
I believe the variable name is rotor_z set in fast physics engine but I need to double check

lukaszdworako, 2017-06-21:
Not in fast physics engine, I would double check in wherever the rotor parameters are for the previously mentioned variable

limzhanfeng, 2017-06-21:
I found rotor_z as one of the local parameters in initializeRotors, part of MultiRotorParams.hpp. However I can't find where initializeRotors is being called.

limzhanfeng, 2017-06-21:
Rotor_z is a real_T, not a vector. It seems that it describes the height of the rotors above/below the COG of the copter rather than the directions which they point in.

lukaszdworako, 2017-06-23:
My apologies the variable is located in MultiRotorParams.hpp and is named unit_z. It follows the XYZ plane and is going to be which direction the thrust is pointed. Again sorry about the confusion.

sytelus, 2017-06-26:
Here is the relevant line:

https://github.com/Microsoft/AirSim/blob/master/AirLib/include/vehicles/MultiRotorParams.hpp#L130

closed 2017-06-26 by sytelus