microsoft/AirSim #299: Question: What are the columns in the text file that is generated after clicking "rec" in Airsim?

vatsagandhi, 2017-06-26:


saihv, 2017-06-26:
See https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/RecordingThread.cpp#L73.

The columns contain timestamp, position and orientation of the drone.

vatsagandhi, 2017-06-26:
Thank you for the quick response

vatsagandhi, 2017-06-26:
Sorry is there a way to reduce the frequency of the recording thread and add for ex. lat,lon,alt into the text file as well?

saihv, 2017-06-27:
I haven't tried it, but it should be possible. AirLib has the method [getGpsLocation()](https://github.com/Microsoft/AirSim/blob/681b79d6946fdb5be141e835915c42f9b0bfbf47/AirLib/include/controllers/DroneControllerCancelable.hpp#L212) that might be useful to you: although invoking this from the recording thread might mean you'll have to figure out the right call namespace-wise etc. Perhaps the devs can give a better response

closed 2017-06-28 by vatsagandhi