microsoft/AirSim #301: Ubuntu build issues

romixlab, 2017-06-26:
After successful build of UE Editor, AirSim static libraries and UE plugin it fails to load with:
`LogLinux:Warning: dlopen failed: /..../libUE4Editor-AirSim.so: undefined symbol:_ZN3msr6airlib19DroneControllerBase10moveOnPath`. This method exists, but somehow didn't make it there... What could be wrong?

P.S. I had to remove this line from AirSim.Build.cs:
`UEBuildConfiguration.bForceEnableExceptions = true;` since compiler complained about setting an read-only variable

There seems to be a mistake in AirLib/include/controllers/simple_flight/firmware/Stabilizer.hpp on lines 16, 18, 20, should be params instead of params_

And lastly had to move static libraries a bit so that UE plugin would link succesfully:
AirLib/lib/x64/Debug/libMavLinkCom.a -> AirLib/deps/MavLinkCom/lib/libMavLinkCom.a
AirLib/lib/x64/Debug/libAirLib.a -> AirLib/lib/
AirLib/lib/x64/Debug/libAirSim-rpclib -> AirLib/deps/rpclib/lib/librpc.a

romixlab, 2017-06-26:
Almost forgot: I'm using Ubuntu 17.04, clang version 3.9.1-5ubuntu1 (tags/RELEASE_391/rc2), cmake 3.7.2, latest UE, latest AirSim.

sytelus, 2017-06-29:
I fixed clang compile errors and warning yesterday. Please feel free to try again.

flyingsilverfin, 2017-07-06:
@romixlab I've also run into the read-only variable issue and removed it. I subsequently run into this:

`progress 'Compiling C++ source code...' 60%
[4/4] Link libUE4Editor-Test_Unreal4_15-1047.so
/usr/bin/ld: cannot find -lUE4Editor-AirSim
clang: error: linker command failed with exit code 1 (use -v to see invocation)
@progress 'Compiling C++ source code...' 80%
ERROR: UBT ERROR: Failed to produce item: /local/scratch/build/projects/Test_Unreal4_15/Binaries/Linux/libUE4Editor-Test_Unreal4_15-1047.so
`

Did you deal with this as well? If not I'll open a new issue.
cheers!

sytelus, 2017-07-27:
We been busy fixing various issues in Linux build. Now we have new code out that should build and run on Ubuntu 16 with few known issues. Please see [docs](https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md)) for instructions.

closed 2017-07-27 by sytelus