microsoft/AirSim #302: Some kind of error with build.sh

lukaszdworako, 2017-06-26:
I have attempted on two different machines, both produce the same error: 
`3 warnings generated.
[ 90%] Linking CXX executable /Users/Lukasz/desktop/AirSim/cmake/output/bin/AirLibUnitTests
Undefined symbols for architecture x86_64:
  "simple_flight::Mixer::mixerQuadX", referenced from:
      simple_flight::Mixer::getMotorOutput(simple_flight::Controls const&, std::__1::vector<float, std::__1::allocator<float> >&) const in AirLibUnitTests.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: 
*** [/Users/Lukasz/desktop/AirSim/cmake/output/bin/AirLibUnitTests] Error 1
make[1]: 
*** [AirLibUnitTests/CMakeFiles/AirLibUnitTests.dir/all] Error 2
make: 
*** [all] Error 2`, both fail same spot same error. Pulled and retried build.

lukaszdworako, 2017-06-26:
UPDATE: upon looking through commits travis-ci has been throwing this error since approx. the 22nd the earliest occurrence I found was the travis-ci test on this commit: [initial simple_flight](https://github.com/Microsoft/AirSim/commit/1b34838258fcb7501b43cc06f06bacf786fea451)

sytelus, 2017-06-28:
Just fixed it. Thanks for reporting it!

closed 2017-06-28 by sytelus