microsoft/AirSim #303: Drone orientation is very unstable

hngenc, 2017-06-28:
Whenever we arm the drone, it begins jerking about in random directions and usually ends up upside down on the ground. We've used Log Viewer to monitor the Pixhawk's orientation, but the Log Viewer only shows the purple drone turning around endlessly without ever settling on any specific orientation, even when the drone is unarmed and perfectly still on the ground.

Does anyone know why the Pixhawk can't seem to find the correct orientation? And does the API provide some way to calibrate the drone's orientation?

sytelus, 2017-06-28:
Can you please elaborate when does this happen. If PX4 is restarted and is in clean state then I haven't seen this happening. However I have seen this happening in following cases:

1. You switch to another window, frame rate drops and suddenly you see drone losing orientation. Same thing might also happen on under powered machines.

2. Another case where I have seen this happening is when you takeoff, do few things and then land but don't disarm. Then restart AirSim again and you might see this.

Can you please elaborate if you are seeing the issue in above cases? Or is it all the time? Also which firmware version are you using?

PS:
The likely reason for (1) is lower frame rate which causes physics loop to shutter, I think.
The likely reason for (2) is that PX4 is not exactly level on landing and sensor calibration gets messed up.

hngenc, 2017-06-28:
1. I've set my preferences so that the frame rate does not drop when I switch to another window. In general, I get about 90 to 110 FPS in the Blocks environment.

2. I've noticed this as well. Disarming lowers the chance of the Pixhawk messing up, but disarming doesn't always seem to work. In other words, sometimes, I'll land and disarm perfectly and when I start up AirSim again, the drone is again very unstable.

Currently, I'm using the PX4 1.6.0 firmware from @lovettchris's fork: https://github.com/lovettchris/firmware.  However, I've also tried 1.6.3 and 1.4.4, and they all have the same problem.

Is there some way to force the PX4 to start from a "clean slate", as you say? I've tried turning it off and turning it on again, but that doesn't seem to do anything.

viniciusguigo, 2017-06-28:
I've had similar issues depending on where I was adding the "Player Start". I found that it should be located really close to the ground (it will even display a "Bad Size" error on Unreal) to properly takeoff.

hngenc, 2017-06-28:
Thanks for the tip, @viniciusguigo, but, unfortunately, that doesn't seem to make a difference in my case.

sytelus, 2017-06-29:
@hngenc what is the setting you are using to keep FPS same when you switch window?

If I do following I always get clean start in Blocks environment:
1. Press Stop button.
2. Restart the device (I use it in HIL).
3. Press Play button.

I think I have a guess of what is happening overall. When vehicle is sitting on ground there is constant collision and slight bounces. Also, if you land, you may land in orientation slightly different from ground plane. I think this is messing up Pixhawks internal calibration probably. Ideally it shouldn't but probably there is bug somewhere in firmware's landing code detection. I'll implement the ground lock in couple of days and then you can check out if that works.

hngenc, 2017-06-29:
@sytelus To keep the same FPS, I unchecked the "Use Less CPU when in Background" option in the Editor Preferences.

I've tried following your exact steps, but they don't seem to help in my case.  Both Log Viewer and QGroundControl show the drone's orientation spinning around endlessly, no matter how many times I restart the flight controller or the simulation.

The SITL, however, works perfectly, so I really do think that this is a PX4 firmware issue.  It could also be an issue with my hardware, but I've tested on two different Pixhawks from two different manufacturers, and they both showed the same behaviour, so I don't think that this is a hardware defect at play here.

I'll try using HIL again when you've added the ground lock, and I'll update with the results.

sg2, 2017-07-07:
We were having a similar issue with HITL. We found that if we rolled back AirSim to an older version, it went away. (We also use PX4 1.6.0 firmware from @lovettchris's fork.) You could try:
```
git clone https://github.com/Microsoft/AirSim.git
git reset --hard 6458eaf44aff5a802d9953457e06b0734f16d6b9
```

sytelus, 2017-07-09:
It appears firmware versions after 1.5.5 have issues such as takeoff doesn't work any longer. I would suggest to use version 1.5.5 and make sure you reset all parameters to default value and the recalibrate RC and set airframe to HIL Quadcopter X. I've tested this with latest code on multiple Pixhawks and PCs and has worked every time.

closed 2017-07-09 by sytelus

zaddan, 2017-07-12:
I am having the same issue with the latest version of airsim. I tried to use px4 1.5.5 firmware, and as was mentioned "reset all parameters to default values and the recalibrate RC and set airframe to HIL Quadcopter X", but taking off and flying still have issues (i.e the drone flies in random directions).

machenxiang, 2019-03-16:
@zaddan @sytelus @viniciusguigo I also have the same question. I used airsim 1.2.1 .and I have tried PX4 firmware 1.4.4,1.5.5 and 1.8.2. but this question still exists.