microsoft/AirSim #304: Combining HIL with actual flying

andersforslund, 2017-06-28:
Hi!
I'm building a custom octocopter, and I want to do Hardware-In-the-Loop testing. I have a CAD-model of my design, so I will just follow the hexacopter example to get it into AirSim. However, my question is this: How do I combine HIL simulations with actual flying? Do I need different firmware, or will the HIL firmware work with real flying out of the box? What's the recommended way of doing it?

lukaszdworako, 2017-06-28:
[Here is a doc on custom drones. ](https://github.com/Microsoft/AirSim/blob/master/docs/custom_drone.md), [I also made a new tutorial ](https://github.com/lukaszdworako/AirSim/blob/master/Custom%20Vehicle.md) that might be a bit more helpful than the hexacopter tutorial since you have 8 props instead of the six in that tutorial.

sytelus, 2017-06-29:
@lukaszdworako would you be able to send the pull request for your doc? It would be great to add in official repo? Also please feel free to send any other changes you think might be helpful.

lukaszdworako, 2017-06-29:
[Here is the link to the actual pull request, ](https://github.com/Microsoft/AirSim/pull/295) I was going to add some information about mixers however that should be handled in the AirSim plugin and PX4.

andersforslund, 2017-06-30:
Thank you for all your replies!
I'm really looking for something quite simple. I don't need to be connected to AirSim while flying, and I'm not carrying an onboard computer. What I would like is this scenario with my octocopter. When the USB cable is plugged in, the firmware is automatically set to HIL mode, and I can run the AirSim simulator and fly around in the Unreal worlds. When I unplug the USB, and maybe move a toggle on my transmitter, the motors are activated and I can go fly in the real world. I don't want a separate setup and firmware for HIL and flying. Maybe this is a newbie question, but how does this work?

aokholm, 2017-07-04:
@andersforslund. I'm not yet familiar with AirSim, but given the way that the PX4 is setup this is not currently possible. In order to run in HIL mode you need to run a specific setup, typically done through the use of a specific HIL "airframe" https://dev.px4.io/en/airframes/adding_a_new_frame.html. You would need to change the parameters on the PX4 system in between HIL and real world tests.

sytelus, 2017-07-09:
Yes, that's correct. However I think only airframe in QGC requires change. Typically we maintain two identical Pixhawks that only differ by airframe.

closed 2017-07-09 by sytelus

andersforslund, 2017-07-12:
Thanks everybody. I've ordered a second Pixhawk :)

omrtozd, 2020-05-05:
> [Here is a doc on custom drones. ](https://github.com/Microsoft/AirSim/blob/master/docs/custom_drone.md) [I also made a new tutorial ](https://github.com/lukaszdworako/AirSim/blob/master/Custom%20Vehicle.md) that might be a bit more helpful than the hexacopter tutorial since you have 8 props instead of the 6 in that tutorial.

Hi, it looks like the tutorial link is broken. How can I find the tutorial?

lukaszdworako, 2020-05-07:
Hi,

sorry for the delay I am not actively working on this project anymore so I'm not actively paying attention to airsim emails...

You can go to my pull request and look through the file additions history which contains all revisions if my document.

Thanks,
Lukasz