microsoft/AirSim #305: C API can not record image

ShukuiZhang, 2017-06-29:
Hi, is anyone using c api to record images? I previously use python api, which works. But when I switch to C API, I can not get the image. 

The following is the code:
'''
int main()
{
	using namespace std;
	//using namespace msr::airlib;

	msr::airlib::RpcLibClient client;

	int i = 0;	
	while (true)
	{
		std::cout << i << endl;
		auto image = client.getImageForCamera(0, DroneControllerBase::ImageType::Scene); 		
		Vector3r position = client.getPosition();
		Quaternionr quaternion = client.getOrientation();
		Vector3r velocity = client.getVelocity();
		
		std::cout << "vector size: " << image.size() << endl;
		std::cout << "position " << position << endl;
		i++;
		if (i > 200)
			break;
	}
	return 0;
}
'''
In above code, I can successfully get the position information, but image.size() gives me 0 all the time. Anyone has an idea? BTW, do I need to make the DroneServer running at the same time? Thanks!!

sytelus, 2017-06-29:
You don't need DroneServer running. Could you please try pressing key 3 when Unreal window is active and drone is armed? That will enable scene sub-window. Please see if that works. I'll try to replicate this tomorrow.

ShukuiZhang, 2017-06-29:
Thanks for your reply @sytelus . I have double checked that Unreal is active and the scene sub-window is there, but can not get the image.

sytelus, 2017-07-09:
I've made lots of changes to image APIs.  You should try out the new version. In your above code, the change is to instead of calling getImageForCamera, use simGetImage.

closed 2017-07-11 by sytelus