microsoft/AirSim #310: EKF baro hgt timeout in SITL

RodgerLuo, 2017-06-29:
I was trying to set the Unreal with SITL, and I got this warning message below after I started PX4: 

[image]

The DroneShell could not return GPS information. And I set the longitude and latitude information as suggested:

[image]

And my settings.json is as below:

[image]

Please let me know what might be the reason to cause this warning. Thank you!

FabianKratzer, 2017-07-05:
Hi,
I ran into the EKF BARO HGT problem, too.
Without the correct altitude only manual control works properly.

PX4 SITL output:
[image]

In HIL-mode UE4 reports takeoff, freefall, landing, too.

settings.json:
[image]

In older versions of AirSim and PX4 firmware the problem didn't occur.
So I think that recent changes, may on PX4 side, cause this problem.
(see: https://github.com/PX4/Firmware/commits/master)

Another question to the position:
Why is it important to set the PlayerStart to a specific position?
(see: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md)
...I can't feel any difference setting the PlayerStart somewhere else...

sytelus, 2017-07-05:
PX4 firmware after release 1.5.5 has issue. See https://github.com/Microsoft/AirSim/issues/316#issuecomment-313218436.

closed 2017-07-05 by sytelus