microsoft/AirSim #312: Turn off physics engine and Pixhawk

Thusheep, 2017-06-30:
Is it possible to fully turn off the physics engine and also the Pixhawk controller? My goal is to move the drone along a spline path to record imagaes, but when clicking play, the drone just "falls down" towards infinity.

lukaszdworako, 2017-06-30:
[I have attempted to disable their physics engine with no luck, here is a link to my question about it. Please tell me if you have any better luck!](https://github.com/Microsoft/AirSim/issues/242)

sytelus, 2017-07-09:
Yes, you can do so now. Look for "computer vision" in [this doc](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md). This is a special mode which turns of drone dynamics and allows you to move drone by keyboard and APIs so you can collect images.

closed 2017-07-09 by sytelus

nirmita123, 2017-07-17:
@sytelus I followed the instructions mentioned in the doc but even after that my "computer vision " mode didn't activate (Since there was no difference the drone dynamics and dint take any instructions from the key board )
This is how my settings.json file looks like after the changes :

{
  "LocalHostIp": "127.0.0.1",
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "ViewMode": "FlyWithMe",
 "FpvVehicleName": "SimpleFlight",
  "UsageScenario": "ComputerVision",
  "RpcEnabled": true,
  "RosFlight": {
    "RemoteControlID": 0
  },
  "SceneCaptureSettings": {
    "Width": 256,
    "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },
  "DepthCaptureSettings": {
    "Width": 256,
    "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },
  "SegCaptureSettings": {
    "Width": 256,
    "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },
  "SimpleFlight": {
    "RemoteControlID": 0
  },
  "Pixhawk": {
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "QgcHostIp": "127.0.0.1",
    "QgcPort": 14550,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "SitlIp": "127.0.0.1",
    "SitlPort": 14556,
    "UdpIp": "127.0.0.1",
    "UdpPort": 14560,
    "UseSerial": true,
    "VehicleCompID": 1,
    "VehicleSysID": 135
  }
}
It would be great if you could help. 
Thanks a lot

lukaszdworako, 2017-07-17:
@nirmita123 I just tried your settings.json configuration and it started in computer vision mode for me, it may sound stupid but are you sure that you are using the latest build of AirSim in your project?

nirmita123, 2017-07-18:
@lukaszdworako I'm using the latest build of airsim (these changes were made before the last airsim build). Seems like I'm missing out a trivial step .