microsoft/AirSim #313: Compiling error in `VehiclePawnBase.h`

BossKwei, 2017-06-30:
Hi All,
I have attempted on running this project on Linux, but I got some problems in the end.
Here is my step-by-step operations:

1. Clone and build Unreal Engine
> Finish without error
2. Clone and build AirSim
> Finish without error
3. Download the template `SunTemple` and follow tutorial `unreal_custenv.md`, then copy the project to Linux
> Finish
4. Copy `AirSim/Unreal/Plugins` to `UE4/Engine/Plugins`
> I noticed that `AirSim/Unreal/Plugins` did not have a folder named `Binaries` like other plugins.
5.  `export EIGEN_ROOT="~/Downloads/AirSim/eigen"`
     `./GenerateProjectFiles.sh -project='~/Documents/SunTemple/SunTemple.uproject'`
> No error
6. `./Engine/Binaries/Linux/UE4Editor ~/Documents/SunTemple/SunTemple.uproject`
> The Editor point out I missed a file named `UE4Editor-AirSim.so`, I clicked Yes then an error has occurred: 
```
LogCompile: .../Engine/Plugins/AirSim/Source/VehiclePawnBase.h(89): Error: Property is exposed to the editor or blueprints but has no Category specified.
Error: Property is exposed to the editor or blueprints but has no Category specified.
```

Is this a bug or I made something wrong? And help will be appreciated.

Ubuntu 16.04
Unreal Engine 4.15.3
Clang 3.9
AirSim master branch now

lukaszdworako, 2017-06-30:
Currently have the same missing module, if some member of the team could advise since many people have experienced this?

BossKwei, 2017-07-01:
@lukaszdworako 
A temporary fix is changing `UPROPERTY(VisibleAnywhere)` to `UPROPERTY(VisibleAnywhere, Category = "Basic")`

But it may be cause other problems, I'm not familiar with UE4, so let's wait team members to solve this problem. ^_^

BossKwei, 2017-07-01:
Hi @sytelus ,
I have solved several bugs and make the AirSim work on Linux, my UE version is 4.15.3, and I noticed that your env may be 4.15.0, I'm not sure if it works properly in your env.
So should I commit my patches(no more than 10 lines) to master branch?

rmst, 2017-07-03:
@BossKwei, I'd be interested in your patches! Could you make your branch available as a fork?

sytelus, 2017-07-09:
@BossKwei Yes! Please send the PR.  Thanks.

closed 2017-07-14 by bosskwei