microsoft/AirSim #314: How can I get camera intrinsic parameters?

Nickvot, 2017-06-30:
Hi, I'm trying to find camera intrinsic matrix.
In the depth map, closer is brighter, so it may be a disparity map I think.
I read some issues that I can find it in point_cloud.py, and the Q matrix in that file is 
```
projectionMatrix = np.array([[-0.501202762, 0.000000000, 0.000000000, 0.000000000],
                              [0.000000000, -0.501202762, 0.000000000, 0.000000000],
                              [0.000000000, 0.000000000, 10.00000000, 100.00000000],
                              [0.000000000, 0.000000000, -10.0000000, 0.000000000]])
...
Image3D = cv2.reprojectImageTo3D(gray, projectionMatrix)
```
...yeah.
However, the format of Q must be like below,

```
Q=[[1, 0, 0, -cx],
      [0, 1, 0, -cy],
      [0, 0, 0, f],
      [0, 0, -1/B, (cx - cx')/B]]
```
Where cx' is cx of the right cam, B is the baseline.
How can I get the f, cx, cy, and the B?

sytelus, 2017-07-09:
You need to first convert depth in to camera plan and then build matrix using camera FOV = 90. See the [discussion here](https://github.com/unrealcv/unrealcv/issues/14).

closed 2017-07-09 by sytelus

badabook, 2018-01-30:
k

stavBodik, 2019-04-22:
@Nickvot

CameraMatrix = IntrinsicMatrix * ExtrinsicMatrix

Where ExtrinsicMatrix = RotationMatrix*TranslationMatrix(aKa position)

For C++ users :

`ProjectionMatrix projMatrix = client.simGetCameraInfo("0").proj_mat;
VectorMath::Matrix3x3f rotationMatrix = client.simGetCameraInfo("0").pose.orientation.toRotationMatrix();
VectorMath::Vector3f cameraPosition = client.simGetCameraInfo("0").pose.position;
 `

orangelx, 2019-12-09:
Did you solve this problem?  like this  #1198

stavBodik, 2019-12-09:
@orangelx 

Please read here :

https://github.com/unrealcv/unrealcv/issues/14#issuecomment-487346581