microsoft/AirSim #316: Unable to use DroneShell nor QGroundControl to control drone

will-kode, 2017-07-03:
Hey everyone,

I am experiencing a few issues with my AirSim environment. But first, here is a brief description of my environment.

CPU: Intel Core i7-6700 @ 3.40GHz
GPU: Nvidia GeForce 980 GX Ti
Memory: 16GB
Storage: ~750GB
Windows 10

I cannot control the virtual drone with the DroneShell application nor the QGroundControl application. Whenever I type the "arm" and "takeoff" commands in DroneShell, the Unreal in-game debugger states that the drone has been armed, but the takeoff command was rejected and that I should disarm and re-try again. However, even after entering the "disarm", "arm", and "takeoff" commands multiple times, I still receive the same message. In the DroneShell console, I receive a "RPC Error" whenever I try to send the "takeoff" command.
[image] In addition, I read that the drone's propellers are supposed to spin once the drone is armed, but they do not.

I also tried using the QGroundControl application to control the drone (as per a suggestion in Issue #96). But my QGroundControl application does not seem to connect or receive messages from AirSim.

Below are some other things that I have noticed that may be related to the issue I am having:
1) After I press the play button, the collision count continually increases even though the drone is not moving.
2) The PX4 SITL console repeatedly states that it detected a takeoff, freefall, and landing. After some time I also repreatedly get "EFK baro hgt timeout - reset to baro" warning. [image]

Any suggestions on how I can fix these issues?

Thank you for the help in advance.

FabianKratzer, 2017-07-05:
#310

sytelus, 2017-07-05:
I can reproduce this issue and have tracked it down to latest PX4 firmware releases. All PX4 releases after 1.5.5 has [this issue](https://github.com/PX4/Firmware/issues/7516). You will be able to arm the drone but take off. I would suggest use [PX4 release 1.5.5](https://github.com/PX4/Firmware/releases/tag/v1.5.5). If you still have problem, try resetting all parameters to default in QGroundControl and then re-calibrate RC and set airframe to HIL QuadcopterX.

will-kode, 2017-07-05:
Thanks for the responses!
I switched to PX4 version 1.5.5 and I no longer receive the RPC error whenever I type "takeoff" into the DroneShell prompt.

However, the drone still does not arm whenever I type the "arm" command nor does it takeoff. I tried typing disarm, arm, and takeoff multiple times, but that still did not work.

@FabianKratzer, in Issue #310 which version of AirSim are you referring to when you mentioned "older versions of AirSim"?

FabianKratzer, 2017-07-06:
AirSim:
Commit 6458eaf44aff5a802d9953457e06b0734f16d6b9
PX4_Firmware:
Commit a55f97503e008b72e22450c58a42002eaf1d9ef3

sytelus, 2017-07-09:
Did you tried resetting the parameters in QGC (this would need re-calibrating RC and setting up airframe to HIL QuadrotorX)?

will-kode, 2017-07-11:
Thanks everyone for your relies.
@FabianKratzer I installed the AirSim and PX4_Firmware commits you mentioned and the simulation no longer erroneously detects takeoffs, landings, and collisions like before (which is actually a good thing). And @sytelus QGroundControl now also receives data from AirSim. However, whenever I use DroneShell or QGroundControl to arm or to takeoff nothing happens in AirSim and, if I am using DroneShell, the prompt in DroneShell says the RPC failed because the drone was not within the takeoff altitude limit.

I've looked at other AirSim Issues to solve this problem, but those that I have tried did not work. Do you all have suggestions?

Btw, I using the SITL, not the hardware version.

Thanks.

sytelus, 2017-08-14:
There have been several changes during last month so you might want to try out the latest. Specifically, ground lock has been re-enabled + baro noise issue has also been fixed.

closed 2017-08-14 by sytelus