microsoft/AirSim #317: Time stamps when "FpvRenderTarget" and "DepthRenderTarget" frames are updated

ShukuiZhang, 2017-07-03:
To make data sets, I need to record the scene and depth images. 
The problem now is either the record button or APIs actually sample the images at some time instant, which means that the time I got for these images is not when they are firstly produced. 
Instead I need the time instant that they are created, in other words, I want to know the time stamps once the frame of "FpvRenderTarget" and "DepthRenderTarget" are updated. Can it be done? Anyone knows where is the line to work with? Thank you in advance!

sytelus, 2017-07-05:
I'm rewriting the camera system and I would incorporporate this suggestion. Hopefully, changes should be out in 2-3 days.

ShukuiZhang, 2017-07-05:
Thank You so much! @sytelus  If it happens, it is truly life saver! On the other hand, I assume the same issue for poses(e.g. orientation, position) will be updated too.

ShukuiZhang, 2017-07-10:
Thanks so much! @sytelus Is it possible to also make the velocity in the return struct of simGetImages()? Sorry that I didn't mention that earlier, but it is important to write our own simulator for imu. Thanks.

sytelus, 2017-07-11:
I've implemented [simGetImages](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md) that returnes camera pose and timestamp. I'm planning to implement API to retrieve IMU. Are you looking for actual IMU sensor readings (which could be noisy) or ground truth angular/linear kinematics? 

I would also caution against using this information to write your own controller. The reason is that controllers for quadrotors needs to operate at fairly high frequency consistently (I would say 100Hz is starting point). I'm not sure if RPC APIs can operate at that speed consistently...

ShukuiZhang, 2017-07-11:
Hi, @sytelus . The reason that I want the ground truth velocity is to generate our own imu ground truth angular/linear kinematics, together with camera poses. 

If you are also thinking about creating the API, I need the ground truth imu kinematics at least, then the measurements can be generated according to different models, you can also provide the real readings as an option.

I'm NOT using this information to write the controller. For now, we need the simulated IMU readings to create a corresponding Data set. Thanks again for taking care of my concern!

sytelus, 2017-07-17:
ok... I'm planning to revamp state APIs. I'll keep this in mind (probably in next 2-4 weeks).

sytelus, 2018-02-13:
We have added new APIs to get angular velocity (along with other kinematics quantities). Please see https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#getmultirotorstate.

For cars, this is available in [getCarState API](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#apis-for-car).

closed 2018-02-13 by sytelus