microsoft/AirSim #322: getImageForCamera Function

yti1993, 2017-07-06:
I got image info that was captured at Unreal-Engine world by using "AirSim/PythonClient/camera.py".  

## My process summary
**First**, I execute Unreal Engine & Airsim.

**Second**, I add  `print (rawimage[0:1000])` at "camera.py" ,  [this line](https://github.com/Microsoft/AirSim/blob/master/PythonClient/camera.py#L45) because I want to get Depth information.

**Third**, I execute "camera.py" on cmd screen (Windows 10)

**finally**, I got matrix about depth_image on cmd screen. 

I watch matrix that is filled many numbers. that matrix and captured Depth image are below.
[image]
[image]

I wonder that matrix numbers what means.... 
they are Depth_info?
they are Disparity_info?

p.s I slice matrix [0:1000], because this image matrix size is very big, huge. so data is omitted.

ShukuiZhang, 2017-07-06:
I'm pretty sure they are disparity images coming from stereo cameras. @yti1993

yti1993, 2017-07-07:
@ShukuiZhang 
Thank you for comment!

I guess.. if that numbers mean 'disparity', range of numbers was 0~255.
but, values on screen are not...

Do you share some evidence of your thinking _"they are disparity images"_  to me?

Have a nice day @ShukuiZhang !

ShukuiZhang, 2017-07-07:
Because the closer the point, the brighter (larger pixel value) it is in the image, which can not be a depth map. @yti1993  I don't know why the numbers are not 0~255, in my case it is. The disparity image comes in rgb (3 channel but same value for each channel), I'm not sure if this is your problem.

sytelus, 2017-11-07:
The disparity and depth image generation had major overhaul: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md

closed 2017-11-07 by sytelus