microsoft/AirSim #323: Anyone can confirm Camera Drone relative pose?

ShukuiZhang, 2017-07-06:
I've been searching for the drone and camera relative pose for a while, which is useful to create my data set. I find a line of code [here](https://github.com/Microsoft/AirSim/blob/92a3f516cb1d72c75b711853839c906c9e290f38/Unreal/Plugins/AirSim/Source/SimMode/SimModeWorldMultiRotor.cpp#L47). I think this is the actual relative pose definition between camera and the drone, can anyone confirm that? @lovettchris @sytelus From this line, I can infer that they are of same orientation and a relative position of [-300, 0, 200]. Am I correct? Is the unit for relative position mm? Thanks!

sytelus, 2017-07-07:
Relative orientation is unity quaternion (i.e. same for both). Relative pose for right camera is x = 46cm, y = -7cm and for left its right camera is x = 46cm, y = +7cm. You can view this by opening up FlyingPawn blueprint in Unreal editor. BTW, I'm revamping camera system as we speak :). I'll update this thread with more info after I'm done.

ShukuiZhang, 2017-07-07:
That is really great! @sytelus So the baseline for the stereo is simply 14cm. Things look much better now.

closed 2017-07-07 by ShukuiZhang