microsoft/AirSim #324: losing altitude when yawing

abdelrahman-t, 2017-07-06:
Is there a way to enable altitude hold through sitl or any otherway while in offBoard mode as i found out tham am losing quite a bit of altitude when yawing, which eventually causes unintentional landing!

i have been going through px4 dev guide to find a solution but to no avail, have any one here tried this ?

thanks

lovettchris, 2017-07-06:
We use "moveByVelocityZ" for this purpose.  But this is currently not an option for moveByAttitude.  Or are you using rotate methods?  You could rotate using moveByVelocityZ by passing different YawModes.

Turns out that real drones lose altitude when yawing counter clockwise, and gain altitude when yawing clockwise, so if the simulator is doing that then it is being ultra realistic :-)

abdelrahman-t, 2017-07-06:
i see, simulator is pretty realistic, i was pulling my hair out, because my reinforcement agent kept doing these unintentional landings instead of yawing.

that's pretty interesting,

thank you chris 😄

abdelrahman-t, 2017-07-06:
on the same not, i was wondering, is there a way to always force the quadcopter to move with camera always pointing in the direction of movment , instead of yawing whilst moving in forwardOnly mode

sytelus, 2017-07-09:
@abdelrahman-t forwardonly mode allows movement only in forward direction.  Please start new issue if that's not what you want.

closed 2017-07-09 by sytelus