microsoft/AirSim #328: expressing velocity so to preserve heading in move by velocity

abdelrahman-t, 2017-07-08:
how can i force the quadcopter to move in the direction it's currently pointing at as opposed to moving while yawing (forward only mode), i understand that vx, vy, are defined with resepect to earth ?, but when the quadcopter frame, and earth's frame are not aligned how can i express vx, vy as to move in the direction the quadcopter is currently pointing at ?

andre-nguyen, 2017-07-09:
you rotate the velocity vector using the quad's yaw...

sytelus, 2017-07-09:
You can transform your velocity vector from body frame to world frame. There is a getOrientation() method that gives quaternion which you can use for transformation. There is a helper method in `[VectorMath::transformToWorldFrame](https://github.com/Microsoft/AirSim/blob/master/AirLib/include/common/VectorMath.hpp#L178)` that we use for this all the time.

abdelrahman-t, 2017-07-09:
@sytelus could you please elaborate a little ? , i mean a velocity vector is in the form [vz, vx, vy]
to command a horizontal velocity in the current direction multirotor is pointing at i should set vz=0, vy=0
and then rotate that vector with the quaternion i get from getOrientation to get the vx,vy with respect to earth , right ?

i did that but quadcopter is still not behaving as expected, am i doing something wrong ?

thanks

closed 2017-08-14 by sytelus