microsoft/AirSim #330: Unreal engine simulation crash in SITL mode on Mac and on Windows

Natalybr, 2017-07-10:
I have tried running AirSim in SITL mode both in MacOS and Windows 10, with no RC, UE4 release 4.15.3. I'm using AirSim-master for windows and a few weeks older version for Mac (I had trouble building the newest version on Mac so I just kept using the older one).

In both cases I tried using the block environment (following [these](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md) instructions) and a simple custom environment (following [these](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md) instructions), with a settings.json file that looks like [this](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md), except I changed useSerial to false. 

- In MacOS, when I hit play in the Unreal Engine (after switching to AirSimGameMode), it gets stuck and I have to force-quit it, though the log output does not indicate anything is wrong.  

- In Windows, after I hit play it first shows the errors: 
```
Vehicle controller could not be started, please check your settings.json
Could not find a connected px4 flight controller
```
and then crashes. 

In both MacOS and Windows, the PX4 terminal is showing no messages after 
`INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..`.

Please let me know what am I missing here.

sytelus, 2017-07-11:
I'd seen this issue report before where AirSim wasn't able to connect to SITL window on Linux. To debug that you can probably use MavLinkTest (see [logging](https://github.com/Microsoft/AirSim/blob/master/docs/logging.md)). If you have xcode or Linux debugging experience, this is good place to contribute :). On Windows using PX4 SITL requires latest version of BashOnWindows. I think you need to upgrade to Windows 10 Creators Edition because older versions didn't had support for ports.

Natalybr, 2017-07-11:
I upgraded to the latest version of Windows as you suggested, though I still get the same errors.
I also tried using [this](https://github.com/Microsoft/AirSim/issues/40) advice with no luck.

When I tried opening a port from BoW and connected to it via UDP from Windows, that worked fine. I also verified with TCPView that PX4 is connected to port 14560. Any idea what might cause this issue?

sytelus, 2017-07-17:
Could you see if you can connect to SITL via QGroundControl? If you can't then very likely you can't connect using AirSim either.

Natalybr, 2017-07-17:
I'm not sure what you mean by connect to SITL via QGroundControl; I opened QGroundControl, added a Link Configuration of UDP type on the PX4 port, and tried connecting to it. 
The following message was shown:
`Connect not allowed during Firmware Upgrade`

sytelus, 2017-07-19:
So may be that's the reason. If QGC can't connect then AirSim can't connect either. Do you see this message every time? Even if you restart everything?

closed 2017-08-14 by sytelus