microsoft/AirSim #331: Randomly Spawn Drone

RodgerLuo, 2017-07-11:
What is the best pipeline to have the drone spawn in the air while the simulation is running? For example, I wonder if I can write a function in Python that once the function is called, the drone will be reset in a random location in the air, armed and hovering. I'm aware that it may involve modifications on Blueprints in Unreal or even PX4 firmware, but it would be great if I can have some instructions. This method would be very helpful for reinforcement learning that the drone can spawn at different locations after episodes end.

sytelus, 2017-07-11:
Are you using PX4? Is so then things are tricky. PX4 takeoff is bit unpredictable and you need to do that before you can hover. Also PX4 doesn't have reset function :(. I'd tried to hack it and its implemented when you press backspace key (which resets to beginning position) but as you can see sometime you have to press Backspace twice.

I'm writing new flight controller called [simple_flight](https://github.com/Microsoft/AirSim/tree/master/AirLib/include/controllers/simple_flight) that would run from inside of AirSim without any external HITL/SITL. This is precisely for the purpose of reinforcement learning. The code for this flight controller would be very simple and easily hackable to whatever functionality one needs. So far I've got rate and level controllers *sort of* working and I wouldn't call it read for use yet. But if you are willing to hack then dive in to that code!

Finally, one other option is to use new "Computer Vision" mode that I just implemented yesterday :). In this mode, you can control it new `simSetPose` API and then use `simGetImage` to get image. In this mode drone acts as "ideal drone". You tell it where to go and it goes there perfectly (unless collision happens). You reinforcement learning algorithm can take image as input and give velocity vector as output. You then compute next position as `p += v * dt` and set that position in vehicle usin `simSetPose`. For more on these APIs, please see [this doc](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md).

RodgerLuo, 2017-07-11:
Thank you very much for the answer. We were trying to use PX4 for our spawn feature, but it seems very complicated to achieve. 

The Simple Flight controller sound great. So if we train our models of reinforcement learning in Simple Flight, then test and refine the models in SITL and  HITL, and then eventually on the real drone, do you think it's a workable pipeline? What would be the physical simulations like in Simple Flight? Ultimately, we'd like to have our policies work on the real drone, so we are afraid that Computer Vision scenario may not work for us.

sytelus, 2017-07-12:
Yes, that's exactly the plan. Due to instability and various issues with PX4 its very hard to work with it for reinforcement learning. My plan is to make simple_flight real firmware and actually installable on boards such as Naze32. Unlike PX4, simple_flight will have only basic attitude controller which is what we need for reinforcement learning. It will also be pure C++ cross platform code that is available as header-only library so one can easily debug and edit.

closed 2017-07-12 by sytelus

flyingsilverfin, 2017-07-14:
Just came across this - simple_flight sounds great! I'm trying to use multiple drones in AirSim at once - I adding such functionality using simple_flight would be easier than attempting to use HITL? If this isn't the right place to ask I could open a new issue :)

sytelus, 2017-07-17:
Yes, simple_flight would work much better for multiple drone scenarios. MavLInk requires opening up port or serial I/O and its not efficient for multiple drone scenario.

gmgremillion, 2017-07-31:
@sytelus You mentioned using backspace to respawn the vehicle back at the player start location/origin, which I'm assuming is the method demonstrated in [Chris' video](https://youtu.be/d_FyjKDWQfc?t=1m50s) (at 2:01). I am able to replicate this myself with a SITL setup. 

How might you send this backspace command programmatically from a python script/API, given that it specifically requires clicking on the Unreal simulation window? I'm using SITL to avoid having to physically restart the pixhawk. @viniciusguigo