microsoft/AirSim #333: Is there any easy way to use MAVROS?

Nickvot, 2017-07-11:
Hi, I'm trying to develop a reinforcement learning system using the AirSim as an environment. I need IMU sensor data as the state, so I'm gonna receive it via MAVROS. I found a post in the "Issuses" and I got someone connected the AirSim and the MAVROS. Can someone share the knowledge how to connect them? Of course there is a tutorial about intercepting MavLink messages, however using MAVROS would make it much more easy to use the AirSim as an environment for RL.

KeeganVG, 2017-07-28:
Did you by chance ever figure this out? I'd be interested if you did.

sytelus, 2017-08-29:
simple_flight is available now and is default. This eliminates need to setup PX4.

Closing this as dup of https://github.com/Microsoft/AirSim/issues/53

closed 2017-08-29 by sytelus