microsoft/AirSim #337: New Airsim Commit Is Not Building

exploke, 2017-07-12:
The latest commit of Airsim is giving me errors when I attempt to build the solution. I have reinstalled Airsim several times with the same results. Using the Airsim commit prior to the latest commit is working perfectly for me.

Here are the errors I am receiving.
[image]

lukaszdworako, 2017-07-12:
Appears as though their "Fix for Linux" has caused some kind of error for windows. @sytelus can you advise as to what may have gone wrong?

sytelus, 2017-07-17:
Could you please try the latest code? I just tested in Windows and seems to work fine.

closed 2017-07-17 by sytelus