microsoft/AirSim #338: Ubuntu crash on play button

lukaszdworako, 2017-07-12:
Currently whenever I press the play button I receive a sigsegv fault, same issue as #176 and #272 any advice from devs in order to remedy this issue. I have tried what was mentioned but it did not help. [Here is a link to the crash log.](https://drive.google.com/file/d/0B2jqMoc9y9baUmhmOVl6VTZiVk0/view?usp=sharing)

sytelus, 2017-07-17:
Please try latest code. I just tested Linux end-to-end install and it works fine with latest code. There were several issues resolved last week. To simplify debugging, I would also suggest to use new "Computer Vision" mode [described here](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md). This basically requires adding below in your settings.json:
```
{
"FpvVehicleName": "SimpleFlight",
  "UsageScenario": "ComputerVision"
}
```

If you still get crash, look in to the Saved folder in Unreal project. You should see folder named "Crashes" somewhere and a log file which would have error message at the very end.

lukaszdworako, 2017-07-17:
Newest build works. Thank you.

closed 2017-07-17 by lukaszdworako