microsoft/AirSim #341: How to control UAV in SILT?

fanguu, 2017-07-13:
I follow the guide and want to control the UAV in SILT. My coumpter is win10 ,i7-6700k and gtx1080.  The setting is
[image]
When I enter the play button, the pix4 gives the information as follow.
[image]
Moreover, the UAV can't move in the UE4.15 as shown follow.
[image]
So I want to ask how to take off or move the UAV.

exploke, 2017-07-13:
I moved the the LocalHostIP line to above the LogViewerHostIP. That seemed to work for me.

[image]

fanguu, 2017-07-14:
I changed the LogViewerHostIP, but the UAV also can't mvoe. 
[image]
The gps setting is error??

fanguu, 2017-07-14:
I find my qusetions and restart the UAV program. Now I can control the UAV in Droneshell, but the command is tedious. Can we control the UAV by keboard?

AndrewEj, 2017-07-17:
Hello, how do you solve the problem ?

closed 2017-07-17 by sytelus