microsoft/AirSim #342: Mac OS libc++ library

AndrewEj, 2017-07-13:
I was trying to build without doing this: 
[image]
Then there is a error: Cannot find the settting.cpp file; So I deleted it, and the build went through. But the plugin apparently has some troubles. The airsim cannot be compiled. So I returned to the previous step and tried to do bash getlibcxx.sh; But there is a new trouble coming out, shown below: 
[image]

AndrewEj, 2017-07-17:
I did some modification on getlibcxx.sh and solved the problem. I  commented out the install--.. line. I don't know whether it gonna affects my other procedure or not.

sytelus, 2017-07-17:
Can you try please latest version. Settings.cpp was deleted some time ago but unfortunately CMakeLists wasn't updated which is why you were getting error.

I think you are mixing up two different thing, compiling AirSim and getlibcxx.sh. The later one is independent library.

closed 2017-08-14 by sytelus