microsoft/AirSim #345: how to modify the game scene ?

helxsz, 2017-07-16:
Hi, I am trying to make the drone fly over a car park full with vehicles, how would I be able to add a scene like that ?

lovettchris, 2017-07-16:
Easiest way to do that would be to find an environment that has something similar then modify it.  Like [this one perhaps](https://www.unrealengine.com/marketplace/downtown#&gid=1&pid=10)... seems to have a parking lot, and lots of cars, which you can then copy/paste to fill the parking lot... the Unreal Editor makes that sort of thing very easy to do.

helxsz, 2017-07-18:
Oh, that looks fanatic, I will try that at first. Further questions, when I get the model, how could I fuse the model in the AirSim and would I able to program to control the vehicles on the parking ?

closed 2017-07-23 by helxsz