microsoft/AirSim #346: Unable to control the drone by Droneshell

AndrewEj, 2017-07-17:
My setup :
System: MacOS
i5 CPU @2.00 GHz
8GB RAM
SSD harddrive
RC : No RC
Flight controller : SITL 

I almost set up everything in mac. I can have the drone on UE4; 
[image]

But I get the message from as shown above:
      Not receiving messages from ....

My PX4 SITL control is working; 
[image]

But when I run the droneshell, I got the weird gps signals; 
[image]

I don't know what happened. I cannot control anything. 

I don't have a flight controller. But to my understanding, we are actually using the px4 firmware as the flight controller. So we can connect the drone in the simulation. We can then control the drone by using the Droneshell or python scripts.  Is my understanding correct or not? 

Please help！！

AndrewEj, 2017-07-17:
I originally have the issue with freefall repeatedly. But I switched back to the previous commit and solved it. I still have the problem above.

AndrewEj, 2017-07-17:
[image]
This warning appears couple of times.

sytelus, 2017-07-17:
What is PX4 version are you using? What is Unreal version? Are you using latest code? I made quite a few Linux related fixes last week so you definitely want to work from the latest. One thing I noticed last week is that there were random errors in C++ stdlib which I'm guessing was because of some mismatch in libc++ that we are asking people to use vs what Unreal uses internally. I'm still debugging that part.

AndrewEj, 2017-07-17:
Thanks for reply. I'm using PX4 from this commitment: a55f97503e008b72e22450c58a42002eaf1d9ef3 ;  My UE 4 is 4.15.3.  How can I test the Mavlink with QGC ? I have another stupid question regarding the setup. Do we have to have a RC to enable the SITL node ??

sytelus, 2017-07-17:
You definitely want to use PX4 1.5.5 stable release. Later versions have problem.

Check section "Interfacing SITL with qgroundcontrol" in [this article](https://404warehouse.net/2016/07/11/px4-software-in-the-loopsitl-simulation-on-gazebo/) to test your SITL setup.

closed 2017-07-17 by sytelus

AndrewEj, 2017-07-17:
I thought I could everything right. I can see the drone shown up on my QGC. But when I use the takeoff order, the drone in UE4 doesn't move at all. Do you know what's gonna be the possible reason ?