microsoft/AirSim #347: MavLinkCom subscriber id may give invalid ids

ditlef, 2017-07-17:
The id is: static_cast<int>(listeners.size() + 1)

Subscibe two listeners:
   => L1 = 1, L2 = 2
 Unsubscribe L1.
   => L2 is invalid.
Add Subsciption:
    => L3 = 2 (L2 = 2)

sytelus, 2017-07-17:
Thanks for reporting the issue. Looks like bug is in [this line](https://github.com/Microsoft/AirSim/blob/master/MavLinkCom/src/impl/MavLinkConnectionImpl.cpp#L326). We need to use C++ remove-erase idiom. Please feel free to make a local change and see if that works.

ditlef, 2017-07-17:
The problem is using the size of the listeners collection as the ID.

I made a quick fix that will work until allocating 4Giga listeners.
A more solid fix is ensuring unique number:
Maybe:  set<int> listenersIds;
   allocate: Find int (1..) not in set. Add to set.
   release: Remove from set.

My fix:
     private: int nextListenerId = 1;

int MavLinkConnectionImpl::subscribe(MessageHandler handler)
 {
-    MessageHandlerEntry entry = { static_cast<int>(listeners.size() + 1),
handler = handler };
+    MessageHandlerEntry entry = { static_cast<int>(nextListenerId++),
handler = handler };