microsoft/AirSim #348: Opening Blocks.uproject on Linux

Missyinvisible, 2017-07-17:
I am trying to open the Blocks.uproject on my Linux machine, however, I always get a dialogue box saying that there are too many errors and suggest that I "Try rebuilding from source manually". The only thing (that I can think of) that may be causing the problem is that I am running Ubuntu 14.04. 

Any suggestions would be really appreciated! 

Thank you!

Mayankm96, 2017-07-18:
I am trying to run the `Blocks.uproject` in UE4Editor in Ubuntu 14.04. I am getting a similar error: `EEOR: UBT ERROR: Failed to produce item: /home/user/Airsim/Unreal/Environments/Blocks 4.16/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so`.

Mayankm96, 2017-07-18:
I built the AirSim again, I think there were a few changes from last night and tried to run the Blocks.uproject. Saying yes to the rebuild option for AirSim.so files, I pressed play in the simulator. However, it crashed with an error similar to what I found when I ran the UE4Editor earlier on Linux.

```
[2017.07.18-15.53.45:720][  0]LogUObjectArray: 146 objects as part of root set at end of initial load.
[2017.07.18-15.53.45:720][  0]LogUObjectAllocator: 24112 out of 0 bytes used by permanent object pool.
[2017.07.18-15.53.45:720][  0]LogUObjectArray: CloseDisregardForGC: 0/0 objects in disregard for GC pool
[2017.07.18-15.53.45:720][  0]LogInit: Using OS detected language ().
[2017.07.18-15.53.45:720][  0]LogInit: Using OS detected locale ().
[2017.07.18-15.53.45:721][  0]LogTextLocalizationManager: No localization for '' exists, so 'en' will be used for the language.
[2017.07.18-15.53.45:722][  0]LogTextLocalizationManager: No localization for '' exists, so 'en' will be used for the locale.
[2017.07.18-15.53.45:757][  0]CrashReportClientLog: CrashReportClientVersion=1.0
[2017.07.18-15.53.45:757][  0]CrashReportClientLog: CrashReportReceiver disabled
[2017.07.18-15.53.45:757][  0]CrashReportClientLog: DataRouterUrl: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data
[2017.07.18-15.53.45:758][  0]LogExit: Preparing to exit.
[2017.07.18-15.53.45:758][  0]LogObj: Freed 0b from 0 cluster array pools.
[2017.07.18-15.53.45:758][  0]LogExit: Object subsystem successfully closed.
[2017.07.18-15.53.45:758][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.07.18-15.53.45:758][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 6 for the default handler. Good bye.
[2017.07.18-15.53.46:922][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2017.07.18-15.53.46:923][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2017.07.18-15.53.46:933][  0]LogExit: Preparing to exit.
[2017.07.18-15.53.46:948][  0]LogExit: Exiting.
```

Missyinvisible, 2017-07-18:
Which version of Clang are you using? I noticed that AirSim specifies 3.9 (interestingly enough the script just runs whichever version you have set to default though) and Epic Games suggest using either 3.3 or 3.5 if you're running Ubuntu 14.04.

I tried it with both clang 3.9 and 3.5, but unfortunately got the same error.

Mayankm96, 2017-07-18:
I used clang 3.9 for it.

ajberchek, 2017-07-18:
I was able to get the blocks project working on Ubuntu 14.04 with the following steps:

* Compile the AirSim plugin
     * On the most recent branch 
     * clang-3.5 and clang++-3.5 are installed and are the clang alternatives 
     * Then recursively copy Unreal/Plugins to Unreal/Environments/Blocks as well as AirLib to Unreal/Environments/Blocks/Plugins/AirSim/Source

* Follow the Linux build guide for unreal engine.
     * Unreal engine branch 4.15 is being used
     * clang-3.5 and clang++-3.5 are installed and are the clang alternatives
     * The list of dependencies in the 14.04 section are installed

* Ensure other necessary dependencies are present
     * Run glxinfo | grep "OpenGL version" to ensure you have opengl 3 or higher, if it suggests installing mesa-utils, do so and you will then have opengl.

* After the above is complete, do the following while clang-3.5 and clang++-3.5 are still the alternatives.
     * Navigate to the UnrealEngine repository
     * Run ./GenerateProjectFiles.sh "\<path to Blocks.uproject\>" with \<path to Blocks.uproject\> replaced by the full path to the Blocks.uproject file
     * Run ./Engine/Binaries/Linux/UE4Editor -opengl3 "\<path to Blocks.uproject\>"
          * This may result in a fatal error proclaiming (in a white colored font) that your ShaderCompileWorker is not present. To fix this, run ./Engine/Build/BatchFiles/Linux/Build.sh Linux Development ShaderCompileWorker -verbose
     * I was then able to select the Blocks.uproject as the project from the GUI, it did some compilation, and launched the Blocks environment, although the quadrotor was not present.

Mayankm96, 2017-07-18:
@ajberchek Well the quadrotor is spawned when you press the play button. It is at that instant that the UE4Editor is crashing.

Mayankm96, 2017-07-19:
Well I got the setup working on Ubuntu 16.04 right now and one thing which I noticed that I didn't do for Ubuntu 14.04 was to set up the [`settings.json`](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) file. It is not created automatically and whenever the simulator is run, I think the UE4Editor looks for this file. 

Also I would suggest to see the [ComputerVision mode](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md). To activate it, simply go to `settings.json` file in  ~/Documents/AirSim and make sure following values exist at root level:
```json
{
  "FpvVehicleName": "SimpleFlight",
  "UsageScenario": "ComputerVision"
}
```

Let me know if it works for you all.

sytelus, 2017-07-19:
I did a bunch of refactoring and clean up in build system and tested it out on Ubuntu 16.04.

build instruments are available now: https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md

There is still a crash when press play button if you try to use PX4 (Computer Vision mode works fine). That's debugging for tomorrow...

curantil, 2017-07-22:
I still get the message that libUE4Editor-Blocks.so and libUE4Editor-AirSim.so are for a different engine version following the instructions at https://github.com/Microsoft/AirSim/blob/master/docs/linux_build.md on Ubuntu 16.04.

sytelus, 2017-07-22:
@curantil You might want to run `./clean.sh` to remove any temp files. Also remove EngineVersion like from .uproject if its present.

curantil, 2017-07-22:
@sytelus That doesn't seem to help

Missyinvisible, 2017-07-24:
I used clang 3.5 to build the engine and then clang 3.9 to build the AirSim. Now, I can open the Blocks.uproject with the AirSim plugin. Interestingly enough though, I am running into the same problem many people have where the engine crashes without any errors when they hit the play button.

closed 2017-07-24 by Missyinvisible

vinbo, 2018-06-15:
thanks to all, this kind of useful to me

aravindk2604, 2018-08-21:
@Missyinvisible : thanks for the suggestion. I built the Airsim again with clang 3.9 by following the instructions [here](https://github.com/lukaszdworako/AirSim/blob/master/docs/linux_build.md) and Unreal was not modified (built first with clang 3.7 on my machine). The Blocks environment works for me and I'm able to load a car model and drive around the scene. Using Ubuntu 16.04.5 LTS.
Even though my answer is kinda late by a year, it might help those who had this error now.